mne-tools/mne-python #2988: Evoked topo plot not behaving as per doc string

kambysese, 2016-03-05:
As I understand the docs `evoked.plot_topomap` in either of these cases below should produce plot(s) for 'mags', 'grads', etc. I don't see this behaviour with e.g. nor my data

```
path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'
condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))
times = np.arange(0.05, 0.15, 0.01)
evoked.plot_topomap(times, ch_type=None)
evoked.plot_topomap(times)
```

agramfort, 2016-03-05:
does says:

https://github.com/mne-tools/mne-python/blob/master/mne/evoked.py#L402

"If None, then channels are chosen in the order given above."

it's maybe a bit unclear but it means that 'mag' will be plotted if
available. Otherwise 'grad' and then 'eeg' ...

PR welcome to clarify

jaeilepp, 2016-03-07:
Similar docstring is used in:
```
evoked.plot_topomap
decoding.base.plot_patterns
decoding.base.plot_filters
decoding.csp.plot_patterns
decoding.csp.plot_filters
preprocessing.ica.plot_components
time_frequency.tfr.plot_topomap
viz.plot_ica_components
viz.plot_tfr_topomap
viz.plot_evoked_topomap
viz.plot_epochs_topomap
```
So if this is to be clarified, it should be done in all of these functions.

kambysese, 2016-03-07:
I'll get to it today.

kambysese, 2016-03-08:
@jaeilepp not finding `viz.plot_epochs_topomap`

jaeilepp, 2016-03-08:
> @jaeilepp not finding viz.plot_epochs_topomap

Sorry, I meant ``plot_epochs_psd_topomap``.

agramfort, 2016-03-08:
closed by #2994

closed 2016-03-08 by agramfort