mne-tools/mne-python #2996: BUG: plot_joint should pick channels

kingjr, 2016-03-08:
@jona-sassenhagen 

Currently, plot joint takes all chans to compute the GFP and find the ERF/P peaks. IMO, it should only get the peaks from the the channels it plots:

```
import numpy as np
from mne import EvokedArray, create_info
from mne.viz.utils import _find_peaks

# setup info
n_time, n_chan = 10, 3
chan_names = ['ch_%02i' % chan for chan in range(n_chan)]
chan_types = ['mag', 'mag', 'misc']
info = create_info(chan_names, 1, chan_types)

# setup data
data = np.zeros((n_chan, n_time))
data[0, 0] = 10  # add ERP on first two channels
data[1, 0] = -10
data[2, 5] = 1e3  # add misc response somewhere else
evoked = EvokedArray(data, info, 0)

_find_peaks(evoked, 1)  # return 5, the peak of the misc response
```

jona-sassenhagen, 2016-03-08:
Good catch. Also pinging @jaeilepp if I don't get here in time.

kingjr, 2016-03-08:
If you do a PR , could you externalize the spatial color function, so that
we can use it for home made plotting (with error bar etc) ? Else I can just
copy paste the code.

jona-sassenhagen, 2016-03-08:
>externalize the spatial color function

Sorry, I don't understand what you mean.

dengemann, 2016-03-08:
> Sorry, I don't understand what you mean.

@jona-sassenhagen the spatial colors code would be nice to have in many other context, maybe we can pull it out and expose it somehow as a util in viz. Clearer?

jona-sassenhagen, 2016-03-08:
Ah okay. I'll see if I can get it out.

jona-sassenhagen, 2016-03-10:
@kingjr is this really about plot_joint?

```
evoked.data[0, 10] = 1000
evoked.plot_joint(picks=range(1, 32));
evoked.plot_joint(picks=range(0, 32));
```

Compare.

(plot_joint takes all channels, but first it creates a surrogate object with all non-picked channels dropped.)

Or am I missing something?

kingjr, 2016-03-10:
From what I had seen, `plot_joint` actually calls `plot`, which itself deals with the `_find_peak`. So no, it's not really about `plot_joint`, it's lower.

jona-sassenhagen, 2016-03-10:
Well but `plot_joint` will run `_find_peaks` on only the picked channels, and `_find_peaks` is a private function that doesn't have a picks arg. So I'm still not sure what specifically the problem is?

(To be precise, `plot_joint` calls `_find_peaks` by calling `plot_topomap`.)

kingjr, 2016-03-10:
Do you manage to reproduce the error ?

jona-sassenhagen, 2016-03-10:
You mean that `_find_peaks` also considers misc channels?

Eric89GXL, 2016-03-10:
I would expect the lines above to do as you say, but the idea is that
plot_joint doesn't end up doing an equivalent operation under the hood
(since it does subselect).

Do you get this error with plot_joint, plot, or what? Can you produce a
minimal example that uses the public plotting functions incorrectly? That
would help isolate the problem.

kingjr, 2016-03-10:
Sure

```
import numpy as np
from mne import Evoked, pick_types
from mne.datasets import sample
fname = sample.data_path() + '/MEG/sample/sample_audvis-ave.fif'
evoked = Evoked(fname, 'Left Auditory')
evoked.pick_types(eeg=True, meg=False, eog=False, stim=True)
# reref
evoked.data -= np.tile(evoked.data.mean(1), [len(evoked.times), 1]).T

# clean stim channels
stim = pick_types(evoked.info, stim=True)
evoked.data[stim, :] = 0
evoked.plot_joint(show=False)

# add crazy value on irrelevant channel
time = np.where(evoked.times >= -.100)[0][0]
evoked.data[stim, time] = 1e20
# in principle this crazy value shouldn't change the plot, but the _find_peak
# seems to use all channels
evoked.plot_joint()
```

[image]

Eric89GXL, 2016-03-10:
That does seem to run counter to what Jona said, thanks. Right
@jona-sassenhagen?

BTW instead of de-meaning the data that way, read_evokeds has a baseline
argument you can use.

Eric89GXL, 2016-03-10:
(And even without that, instead of np.tile you could use np.newaxis to
exploit broadcasting, like using bsxfun instead of repeat in MATLAB)

kingjr, 2016-03-10:
Thanks, I knew these tricks existed, but I never remember them on the fly.

jona-sassenhagen, 2016-03-10:
Ah, I see. This is most likely because if picks is empty, _find_peaks gets forwarded the stim channel by evoked.plot_joint.

The same thing goes for evoked.plot_topomap btw (try it).

I'm not sure how to best approach this. Perhaps the easiest option is for _find_peaks to default to only data channels, or, alternatively, for evoked.plot_topomap to default to only data channels if picks is empty. Opinions?

kingjr, 2016-03-10:
IMO the peaks should be set from and only from the plotted channels, whether they are manually or automatically defined.

jona-sassenhagen, 2016-03-10:
I have an idea ... I'll try.

jona-sassenhagen, 2016-03-10:
Okay, I got it. I'll make a PR when #2975 is done, is that okay @kingjr ?

And I'd like to handle factoring out the spatial color code in another PR.

kingjr, 2016-03-10:
Ok, no rush on my side.

closed 2016-03-20 by agramfort

jona-sassenhagen, 2016-07-07:
@kingjr do you still think the spatial_colors cmap should be externalized btw?

dengemann, 2016-07-07:
I liked that idea.

kingjr, 2016-07-08:
no strong opinion.

PR #3003 (2016-03-20): MRG: _find_peaks and _plot_joint picks fix
closes #2996