mne-tools/mne-python #2997: Right singular vector in "label_sign_flip"

YingYang, 2016-03-08:
In line 1082-1085 of mne/label.py

1082:  _, _, Vh = linalg.svd(ori, full_matrices=False)
1083: 
1084: # Comparing to the direction of the first right singular vector
1085: flip = np.sign(np.dot(ori, Vh[:, 0] if len(vertno_sel) > 3 else Vh[0]))

If I am following correctly,  "ori" is a [n_dipoles, 3] vector.  According to "numpy.linalg.svd", the right singular vectors are the rows rather than columns of "Vh".  So should line 1085 be

flip = np.sign(np.dot(ori, Vh[0])) ?

Thanks.

natalieklein, 2016-03-23:
I agree with Ying, and especially do not understand why the code uses the first row Vh[0] when n_dipoles < 3, but uses the first column Vh[:,0] otherwise.

I stepped through the code myself with an example src and label to check... yes, ori has shape [n_dipoles, 3], where n_dipoles is len(vertno_sel) (number of dipoles in the label). According to documentation, numpy.linalg.svd returns Vh = V.H of size [min(n_dipoles, 3), 3]. If we had V, the columns would be the right singular vectors, but we have V.H, so the right singular vectors should be the rows, so we would want Vh[0] no matter what n_dipoles. Unless we are both missing something here... any input or other thoughts would be helpful!

Thanks

agramfort, 2016-03-23:
hum... would what you suggest work in terms of dimensions?

can you try to do the change and see the impact of this example

http://martinos.org/mne/dev/auto_examples/inverse/plot_compute_mne_inverse_epochs_in_label.html#sphx-glr-auto-examples-inverse-plot-compute-mne-inverse-epochs-in-label-py

?

natalieklein, 2016-03-24:
So as Ying said originally, I think it should always be `Vh[0]`, never `Vh[:,0]`.

For the example: by chance, the flip vector comes out the same, but the arguments to `np.sign()` in line 1084 of label.py are **not** equal (ie `np.dot(ori, Vh[0,:]) != np.dot(ori, Vh[0])`)

I changed the label in the example to 'Aud-rh' to see if I could get different behavior, and now the signs indeed come out different. See the attached plot - "myflip" is where I made this change.

[test_signflip.pdf](https://github.com/mne-tools/mne-python/files/187998/test_signflip.pdf)

Eric89GXL, 2016-03-24:
@natalieklein I agree it should always be the row `Vh[0]`. Here is a simple example to illustrate the problem:
```Python
>>> x
array([[ 1.        ,  0.        ,  0.        ],
       [ 0.        ,  1.        ,  0.        ],
       [ 0.        ,  0.        ,  1.        ],
       [ 1.41421356,  1.41421356,  0.        ],
       [ 1.41421356,  1.41421356,  0.        ]])
>>> linalg.svd(x, full_matrices=False)[2]
array([[-0.70710678, -0.70710678, -0.        ],
       [ 0.70710678, -0.70710678, -0.        ],
       [ 0.        ,  0.        ,  1.        ]])

```
The first row is pretty clearly better aligned with the vectors than the first column. @agramfort agree?

Do you want to make a PR? If not, I can do it.

natalieklein, 2016-03-24:
@Eric89GXL I can do it if you want, since I already have the change in my fork. Good catch @YingYang !

Eric89GXL, 2016-03-24:
We'll want to add some test. Maybe:

1. Make a label with 5 arbitrary vertices (can be left-hemi `np.arange(5)`)
2. Load a source space for `sample`
3. Modify the source space for the chosen vertices to have the normals I defined above
4. Verify that `assert_allclose(np.abs(np.dot(flip_vector, [np.sqrt(2), np.sqrt(2), 0])), 1.)`, i.e. it's aligned a reasonable way

That test would fail on `master` and pass on your PR, satisfying TDD.

Eric89GXL, 2016-03-24:
But yes please take a shot at it. If my test suggestion doesn't make any sense I can make a PR into your branch

natalieklein, 2016-03-24:
@Eric89GXL I took a stab at it but I don't understand step 4 (the flip_vector has dimension 5 so it won't dot with dimension 3 so I'm not sure what to do). I committed what I have to my branch in test_label.py, it just needs step 4 of the test.

Eric89GXL, 2016-03-24:
My test code was a bit off, I'm opening a PR with the fixes

YingYang, 2016-03-24:
@natalieklein @Eric89GXL Thanks for testing and fixing it!

PR #3067 (2016-03-24): MRG: Fix sign flip
Ready for review/merge from my end.

Closes #2997.
closed 2016-03-24 by YingYang