mne-tools/mne-python #2998: Bug in equalize_epoch_counts

jaeilepp, 2016-03-09:
It seems the epochs aren't dropped when using categorized epochs.
```
In [64]: len(epochs['standard'])
Out[64]: 387

In [65]: len(epochs['deviant'])
Out[65]: 79

In [66]: mne.epochs.equalize_epoch_counts([epochs['deviant'], epochs['standard']])
Dropped 0 epochs
Dropped 308 epochs

In [67]: len(epochs['standard'])
Out[67]: 387
```
If I store the epochs as their own variables the function works fine.

agramfort, 2016-03-09:
yes because epochs['deviant'] are copies. So you update a copy and then
trash it....

this equalize_epoch_counts turns out to be massively used (beyond decoding)
and I think it's time to consolidate the API.

any suggestion to improve API / naming ?

Eric89GXL, 2016-03-09:
Do you think the name is too long or API too complex? Seems fine to me.

What we really should do is document __getitem___, __add__, etc. for
classes that have them.

agramfort, 2016-03-09:
>
> Do you think the name is too long or API too complex? Seems fine to me.
>

redundance of word epochs, operation is inplace for a function (non-method)

along the same line epochs.drop_epochs is a bit awkward... It should be
epochs.drop or epochs.delete (cf. np.delete)

What we really should do is document __getitem___, __add__, etc. for
> classes that have them.

+1. how to do this?

indeed add and sub etc. don't appear eg in
http://martinos.org/mne/dev/generated/mne.Evoked.html

Eric89GXL, 2016-03-09:
> +1. how to do this?

I tried recently to include these special members using recommendations from [stack exchange] (http://stackoverflow.com/questions/15903577/include-specific-special-methods-in-sphinx) but had no luck. But I must have been doing something wrong.

Even once we get that working, though, we should really add examples to the constructor so that people see those use cases ASAP.

Eric89GXL, 2016-05-11:
@agramfort you still want to change `equalize_epoch_counts` to something else like `equalize_counts`?

Otherwise I'll re-title that we should find some way to properly document `__getitem__` and the like.

agramfort, 2016-05-11:
yes an epochs method should not contain the word epochs

Eric89GXL, 2016-05-11:
Actually we are okay, it's already `epochs.equalize_event_counts` not `epochs.equalize_epoch_counts`. We do have a *function* `equalize_epoch_counts`, but that's fine.

agramfort, 2016-05-12:
having a function (not a method) with inplace updates and no copy param is certainly a bad idea

this is where the fix should be

PR #3224 (2016-05-12): MRG: Document special members
This adds documentation of `__contains__`, `__getitem__`, `__iter__`, and `__len__` to our docs.

Closes #2998.
closed 2016-05-12 by agramfort