mne-tools/mne-python #2999: BUG: inconsistent authors formatting in docs

kingjr, 2016-03-09:
in the examples, the authors are in the script:
[image]

in the tutorials, two font types are used for the authors:
[image]

jasmainak, 2016-03-09:
Fix welcome.

kingjr, 2016-03-09:
I don't master sphinx sorry.

jasmainak, 2016-03-09:
You have to simply move the authors into the comment like this: https://github.com/mne-tools/mne-python/blob/master/tutorials/plot_cluster_methods_tutorial.py#L55

Eric89GXL, 2016-03-09:
@jasmainak does that work for multi-line author lines? It seems like that's the problem here

jasmainak, 2016-03-09:
yep it does. Example here: http://www.martinos.org/mne/stable/auto_tutorials/plot_ica_from_raw.html.

Eric89GXL, 2016-05-11:
This looks fixed to me, feel free to reopen if it's not

closed 2016-05-11 by Eric89GXL