mne-tools/mne-python #3001: MNE report multithreading crashes

kambysese, 2016-03-09:
I've noticed frequent hang-ups during reporting as of late signaled by some cryptic messages related to window management and/or multi-threading processes, possibly related to [this](https://stackoverflow.com/questions/22626989/multithreaded-gstreamer-w-pygtk-crashing-xcb-xlib-threads-sequence-lost)

> [xcb] Unknown sequence number while processing queue
> [xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
> [xcb] Aborting, sorry about that.
> python: ../../src/xcb_io.c:274: poll_for_event: Assertion !xcb_xlib_threads_sequence_lost' failed.
> [xcb] Unknown sequence number while processing queue
> [xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
> [xcb] Aborting, sorry about that.
> python: ../../src/xcb_io.c:274: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.

this is happening in both IDE python console and shell instances. Does anyone know have an idea what this trace means, and better yet how I can make it go away?

jasmainak, 2016-03-09:
Do you also get this error if you set the matplotlib backend to `Agg` ?

```py
import matplotlib
matplotlib.use('Agg')
```
in the python interface

kambysese, 2016-03-09:
I tried that when working in the IDE, it helps to suppress figure windows and error, but things go awry when working remotely. The most recent occurrence though was with usual qt4 backend with python in the shell. Apparently the html file is written out but looks like something gets hung during display management. If I were to guess wildly, it may be related to where n_jobs is set to 4 in the reporting. Is that a recent change?

kambysese, 2016-03-09:
@jasmainak FYI more details using qt4 backend, python in shell, in XRDP session:

> [xcb] Unknown sequence number while processing queue
> [xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
> [xcb] Aborting, sorry about that.
> python: ../../src/xcb_io.c:274: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.
> 312 matching events found
> Created an SSP operator (subspace dimension = 6)
> 312 matching events found
> Created an SSP operator (subspace dimension = 6)
> 6 projection items activated
> [xcb] Unknown sequence number while processing queue
> [xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
> [xcb] Aborting, sorry about that.
> python: ../../src/xcb_io.c:274: poll_for_event: Assertion !xcb_xlib_threads_sequence_lost' failed.
> python: Fatal IO error 11 (Resource temporarily unavailable) on X server :11.0.
> PyCUDA ERROR: The context stack was not empty upon module cleanup.
> A context was still active when the context stack was being
> cleaned up. At this point in our execution, CUDA may already
> have been deinitialized, so there is no way we can finish
> cleanly. The program will be aborted now.
> Use Context.pop() to avoid this problem.

HTML file not written with this. In my experience CUDA/display errors are a nightmare to figure out and even worst to get around, with little to no pay off. I'm ready to give up on this one.

jasmainak, 2016-03-09:
okay, you didn't tell me that you were running it remotely :) Do you have mayavi figures? Have you tried this: https://en.wikipedia.org/wiki/Xvfb ?

jasmainak, 2016-03-09:
also I'm not sure about the pycuda error. Maybe @Eric89GXL has insights

kambysese, 2016-03-09:
@jasmainak no mayavi figures in this instance. At the moment working locally with `Agg` backend. During epoch file processing logger indicating:

> 6 projection items activated
> [Parallel(n_jobs=4)]: Done   5 out of   4 | elapsed:    5.2s remaining:   -1.0s
> [Parallel(n_jobs=4)]: Done   5 out of   4 | elapsed: 63.0min remaining:  -756.2s

63 minutes seems awfully long! Are you using CUDA for anything in `report`?

Eric89GXL, 2016-03-09:
CUDA filtering should not be used in multiple threads. I doubt anyone has
tried using CUDA filtering inside the parallel bits of Report before, so
it's likely we have a bug by not protecting against it.

What commands did you use that ended up causing that error? Can you make a
minimal example with the sample dataset?

Eric89GXL, 2016-11-18:
Closing, feel free to reopen @kambysese if you hit this again (and especially if it can be reproduced on `sample`)

closed 2016-11-18 by Eric89GXL