mne-tools/mne-python #3002: Computing morph matrix on the fly is significantly faster than morph_data

reddigari, 2016-03-09:
Using `mne.compute_morph_matrix` and `mne.morph_data_precomputed` is ~100x faster than `mne.morph_data`.

The relevant morph map already exists and isn't re-written by either command.

**Output:**
```
Data to be morphed:
<SourceEstimate  |  5124 vertices, tmin : -200.0 (ms), tmax : 1200.0 (ms), tstep : 1.0 (ms), data size : 5124 x 1401> 
__________________________________________________
Compute morph matrix and morph_data_precomputed: 0.895 s
Morph_data without precomputed morph matrix: 103.332 s
```

**Script:**
```
stc = mne.read_source_estimate('processed_files/stcs/factorial/no_morph/noun/A0008_noun_no_morph-lh.stc')
fs = mne.read_source_spaces('mri/fsaverage/bem/fsaverage-ico-4-src.fif')
vertices_to = [fs[0]['vertno'], fs[1]['vertno']]

print "Data to be morphed:\n", stc, '\n', 50*'_'

t0 = time.time()
morph_mat = mne.compute_morph_matrix('A0008', 'fsaverage', vertices_from=stc.vertices, vertices_to=vertices_to, subjects_dir='mri')
stc_morphed = mne.morph_data_precomputed('A0008', 'fsaverage', stc, vertices_to, morph_mat)
delta_t = time.time()-t0

print "Compute morph matrix and morph_data_precomputed: %.3f s" %delta_t

t0 = time.time()
stc_morphed = mne.morph_data('A0008', 'fsaverage', stc, grade=vertices_to, subjects_dir='mri')
delta_t = time.time()-t0

print "Morph_data without precomputed morph matrix: %.3f s" %delta_t
```

dengemann, 2016-03-09:
Ok that's expected, where is the problem?

reddigari, 2016-03-09:
[This example](http://martinos.org/mne/stable/auto_examples/inverse/plot_morph_data.html#sphx-glr-auto-examples-inverse-plot-morph-data-py) shows them to be reasonable alternatives to one another. Is it just the size of my source estimate (1400 timepoints) that makes `morph_data` a bad idea?

dengemann, 2016-03-09:
@reddigari there is some heavy number crunching going on under the hood. The example shows that the  output is roughly equivalent. The .morph_data way is expected to be much slower. Then it depends of course on your computational architecture and the data. Ah and yes, 1400 time points clearly make a difference ... If you can, use the precomputed way. It will make you happy.

teonlamont, 2016-03-10:
@dengemann what leads to the difference if the morph_matrix is read from disk, sorry I didn't follow.

Eric89GXL, 2016-03-10:
You either precompute a sparse morph matrix to dot with your data, or you
smooth the data directly. If there are thousands of time points the former
is faster, if there are a few time points the latter should be faster.
Somewhere in between it switches. (Doesn't have anything to do with reading
from disk AFAICT, unless you've saved your morph matrix somewhere.)

We could probably do some benchmarks and figure out which is faster, and if
someone calls morph_data with too many time points, internally switch the
call for them. Personally I just always precompute.

teonlamont, 2016-03-10:
ok, so I understand this all correctly, with `mne.morph_data`, internally it loads a morph_matrix that is saved if found.

but the slow down happens when that matrix needs to be smoothed before being applied to the data.

Eric89GXL, 2016-03-10:
No. The thing that gets created and saved if it didn't exist is a "morph
map" which says which 3 vertices (and weights) in subject b's brain each
vertex in subject a's brain map into. This is not sufficient for actually
morphing data though. This morph map is instead used both by the
morph_precomputed and morph_data routines to smooth data (in mathematically
equivalent but algorithmically different ways) so it's not the
bottleneck/speed difference referred to by this issue. That bottleneck
comes from the differences in those two processes I described before.

Eric89GXL, 2016-03-10:
Oh actually rereading your comment it sounds like you did get it already --
the morph maps midst be processed and how that processing is done leads to
speed differences, yes.

agramfort, 2016-03-10:
we should improve documentation on this: in docstrings with see also and in
the narrative doc...

let's keep this issue open while it's not done.

dengemann, 2016-03-10:
Funny. I read the inverted meaning into this issue title, matching my expectations that precomputing is faster. All what I wrote was following this assumption.

PR #3024 (2016-03-17): MRG: FAQ update
A bunch of touch-ups and additions to the FAQ to get us closer to 0.12.

I also updated our `conf.py` to look like the upstream one from `sphinx-gallery` because otherwise I couldn't build the doc :(

Closes #3021.
Closes #3009.
Closes #3002.
Closes #2945.
Closes #2944.
closed 2016-03-17 by dengemann