mne-tools/mne-python #3006: ENH: combine_evoked could use field mapping to transform data

Eric89GXL, 2016-03-11:
When combining data using `grand_average`, which calls `combine_evoked`, if different subjects (with different device<->head transforms) are used, then the data won't really make sense. Should we add an option to use field mapping code to transform all combined evoked instances to the same dev_head_t?

cc @kambysese

agramfort, 2016-03-11:
does it actually help if head shapes are different?

Eric89GXL, 2016-03-11:
It should at least be better than doing no transform at all. I think it's
useful potentially for infant data where individual head geometry often
isn't known, and also for adult data to get a quick sensor space assessment
of data. WDYT?

agramfort, 2016-03-11:
no opinion.

kambysese, 2016-03-11:
my intuition here is that if rotations are involved in the transformations esp in the case of infants/children then it would be more trouble in terms of distorting the group level field maps than (i) transforms that are only translations vs. (ii) a common transform (that seems ideal). I can try and test this idea for a handful of subjects using different trans scenarios and inspect the resulting topomaps. I assume someone who speaks better linalg-ese than I would have more than a hunch about this :)

jona-sassenhagen, 2016-03-11:
Possibly a similar issue.
A friend of mine had the following issue recently: student assistant had messed up EEG setup. This means that the channel names were in the wrong order for one subject. When averaging, this lead to strange results. My fix was fundamentally easy, but also a bit inelegant. Maybe it would make sense to lookup and match channel names when combining evokeds.

kambysese, 2016-03-12:
in brief for n=18 subjects using an identical `dev_head_t`

[image]

and assuming an identical transform

[image]

Alas this is infant data and I don't really know which I prefer, that said for a decent sample size there is substantial scaling and distortion associated with the choice of transform.

[Here is the gist](https://gist.github.com/kambysese/9370bb619fc12693c409)
btw I don't know why the butterfly plots are generated twice, though there is usually a `matplotlib` warning about the usage of `tight_layout`

agramfort, 2016-03-12:
not sure what you did in details but if this is auditory data the second
topos look better especially for the grads.

kambysese, 2016-03-14:
@agramfort Yes it is auditory (infant) data, and yes I agree that the second topos look more desirable. All the data were processed with movement compensation and svd regularization using `maxwell_filter` twice, (a) using the subjects mean head pos as the `dev_head_t` (b) using `[0, 0, 30]` as the head transformation. Then the data was treated with typical cardiac SSP, and peak-to-peak channel thresholding for trial epoching before averaging. In the case of infant/children data scenario (a) is preferred WRT minimizing reconstruction error SSS denoising & subsequent source recon, whereas (b) is necessary if sensor analyses is the end goal. I suppose if there is an option to use field mapping code to transform all combined evoked instances to the same `dev_head_t`, it would spare users from repeating processing, WDYT?  cc @Eric89GXL

kambysese, 2016-03-16:
@agramfort any further thoughts on this?

agramfort, 2016-03-16:
+1 for allowing to use dev_head_t in grand_average

Eric89GXL, 2016-03-17:
We can adapt the mapping code I wrote in a gist

https://gist.github.com/Eric89GXL/3a44d6b160c9b04a99ad#file-head_trans-py-L96

jasmainak, 2016-03-17:
did we manage to figure out what was wrong with the gist, why the error was so high?

Eric89GXL, 2016-03-17:
I don't remember what the error was

jasmainak, 2016-03-17:
IIRC 7.5 mm when there was no head movement. As I understood from @agramfort , the dipole fitting error should be of the order of 1 mm.

Eric89GXL, 2016-03-17:
Oh the errors with dipole fitting... no, not sure. We could double check
that the C dipole fitting code gives similar results.

kambysese, 2016-03-30:
@Eric89GXL  'm = _map_meg_channels(info_from_meg, info_to_meg, 'accurate', origin=(0., 0., 0.03))' brings up 

> Traceback (most recent call last):
>   File "/home/ktavabi/anaconda2/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 3066, in run_code
>     exec(code_obj, self.user_global_ns, self.user_ns)
>   File "<ipython-input-10-ce5b01244819>", line 8, in <module>
>     m = _map_meg_channels(info_from_meg, info_to_meg, 'accurate', origin=(0., 0., 0.03))
>   File "/home/ktavabi/Projects/mne-python/mne/forward/_field_interpolation.py", line 134, in _map_meg_channels
>     info_to['dev_head_t'], templates)
>   File "/home/ktavabi/Projects/mne-python/mne/forward/_make_forward.py", line 186, in _create_meg_coils
>     _transform_orig_meg_coils(coils, t, do_es=do_es)
>   File "/home/ktavabi/Projects/mne-python/mne/forward/_make_forward.py", line 195, in _transform_orig_meg_coils
>     coil_trans = np.dot(t['trans'], coil['coil_trans_orig'])
> IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices

This is the behaviour with `mne.io.meas_info.Info` dictionaries from personal data and mne-sample data.

larsoner, 2018-04-16:
Closing, can resurrect if someone ends up needing it

closed 2018-04-16 by larsoner

jasmainak, 2018-04-16:
@larsoner do you have the gist still though? So I can point to people who ask me for this functionality

larsoner, 2018-04-16:
Yep, just needed to swap `larsoner` in for my old name `Eric89GXL` in the URL above

jasmainak, 2018-04-16:
Ah, indeed!