mne-tools/mne-python #3017: raw.info / pick.py error from .fif with BIO channels

tntdynamight, 2016-03-16:
Hi, 
This relates to [#694](https://github.com/mne-tools/mne-python/issues/694). 
Using the latest git mne-python, I get an error when 'BIO001' etc channel is included. 
(` raise Exception('Unknown channel type')`)

I tried guessing and adding into pick.py / channel_type function:
`56     elif kind == FIFF.FIFFV_BIO_CH:
 57         return 'misc'`

but then I get: 
`AttributeError: 'BunchConst' object has no attribute 'FIFFV_BIO_CH'`

Here is a small file including the channels:
https://www.dropbox.com/s/i65bvjhlq2kj1b3/BIO_fif_tsss_mc.fif?dl=0

Appreciate any help with this, 
best 
tnt

agramfort, 2016-03-16:
see fix coming at  https://github.com/mne-tools/mne-python/pull/3018

closed 2016-03-17 by agramfort

tntdynamight, 2016-03-18:
Works great, thanks for the quick update!