mne-tools/mne-python #3021: resample versus decimate methods

teonbrooks, 2016-03-16:
I have a question about the resample method for `mne.source_estimate.SourceEstimate`. why do we have resample here and how is it different from the `decimate` method that we have the `Epochs`? Is it generally safe to resample the `SourceEstimate` if it is doing internal filtering?

teonbrooks, 2016-03-16:
cc: @reddigari

Eric89GXL, 2016-03-16:
Decimate doesn't do any filtering, just subselects samples. It's appropriate if you know your data are already low-passed, and doesn't introduce edge artifacts.

Resampling, on the other hand, does apply a low-pass filter of its own before sub-selecting samples. This therefore can introduce edge artifacts.

Make sense? If so, please open a PR to the FAQ page about the differences between decimation in mne-python, and resampling if you have time. If you don't please re-title this issue to be about adding such a section.

teonbrooks, 2016-03-16:
thanks @Eric89GXL. next question would be, should we have the decimate method added to the `SourceEstimate` as well?

Eric89GXL, 2016-03-16:
It's not as safe because for the SourceEstimate objects we have no knowledge of `lowpass`, whereas for `Epochs` we do in `info['lowpass']`.

Eric89GXL, 2016-03-16:
But I think you can get what you want by using `stc.bin`

teonbrooks, 2016-03-16:
ah, I didn't realize that this wasn't in its info. actually, I just discovered that it has no info.

Eric89GXL, 2016-03-16:
@teonbrooks you updating the FAQ section? If not, I can tackle it -- there are a few things I'd like to add.

teonbrooks, 2016-03-16:
I can't at the moment. please feel free to jump in, thanks :)

PR #3024 (2016-03-17): MRG: FAQ update
A bunch of touch-ups and additions to the FAQ to get us closer to 0.12.

I also updated our `conf.py` to look like the upstream one from `sphinx-gallery` because otherwise I couldn't build the doc :(

Closes #3021.
Closes #3009.
Closes #3002.
Closes #2945.
Closes #2944.
closed 2016-03-17 by dengemann