mne-tools/mne-python #3022: Missing .json file in mne package

teonbrooks, 2016-03-16:
The kit2fiff gui now relies on `.json` but this file is not currently included in the `MANIFEST.in`. I am not sure when this change occurred but its inclusion needs to backported to at least v0.11.

cc: @Eric89GXL @christianbrodbeck

teonbrooks, 2016-03-16:
would this just require an additional line something like `mne *.json` and it would propagate through following proper formatting

agramfort, 2016-03-16:
PR to add it to MANIFEST.in welcome.

+1 for backport too.

@jaiellep can you take care of it?

christianbrodbeck, 2016-03-16:
The GUI *.json files are in `setup.py`'s `package_data`. I assume if you put them into MANIFEST.in you can take them out of setup.py? What is the difference?

teonbrooks, 2016-03-17:
I think that with the MANIFEST.in, it tells pip or other packagers what files to include whereas with setup.py, you have the entire package there. I know that one of the users told me that they were having problems (missing `help.json`) when they just installed mne through pip.

closed 2016-03-17 by Eric89GXL

christianbrodbeck, 2016-03-17:
Hm thanks, good to know that setup.py is not enough

agramfort, 2016-03-17:
MANIFEST.in is used to generate the sdist pushed on pypi.