mne-tools/mne-python #3036: Q: parallel in sub function which could be called by parallel

kingjr, 2016-03-18:
I was thinking of parallelizing the [wavelet decomposition functions across wavelet](https://github.com/mne-tools/mne-python/blob/master/mne/time_frequency/tfr.py#L207).

However, these functions [can be called by other function in a parallel mode](https://github.com/mne-tools/mne-python/blob/master/mne/time_frequency/tfr.py#L428).

Such that we'd get something like 
```
def bar():
    ...

def foo():
    parallel(bar())

def joe():
    parallel(foo())
```

Of course we can ensure that the n_jobs is only > 1 at the highest level, but I was wondering whether the overall structure was ok.

dengemann, 2016-03-18:
That cannot work with joblib AFAIK.

kingjr, 2016-03-18:
> That cannot work with joblib AFAIK.

Actually, I just realized that we have are own parallel wrapper, and it doesn't call joblib if n_jobs==1. https://github.com/mne-tools/mne-python/blob/master/mne/parallel.py#L53

So I guess it should be ok, right?

dengemann, 2016-03-18:
Yes that's right. Maybe I misunderstood your question.

kingjr, 2016-03-18:
Ok, I'll wait for https://github.com/mne-tools/mne-python/pull/3034 before trying this, since it's going to be on the same file.

On a related topic, I just saw that [joblib can share objects across workers using memmapping](https://pythonhosted.org/joblib/parallel.html). I've never tried this, any disadvantage (it dumps the data on the hard drive?)?

dengemann, 2016-03-18:
We have memmapping support already in MNE for joblib. My impression was
that it does not make a big diff. What you gain in memory you loose it in
time.

kingjr, 2016-03-18:
The advantage is also for code itself, isn't it? Instead of  chunks of data (e.g. if you loop across chunks of freqs, or chunks of times), you can just load and update and initialized array. It would have made the GAT code much more readable...

closed 2016-05-14 by kingjr