mne-tools/mne-python #3038: Error in create_ecg_epochs with ecg channel: 'Data must be a 2D array of shape'

tntdynamight, 2016-03-19:
Hi, 

create_eog_epochs with method correlation works for me, and 'abusing' it for ecg detection also works, but the 'appropriate' function with 'correlation' method throws an error. 

`ecg_epochs = create_ecg_epochs(raw, tmin=-.5, tmax=.5,ch_name='ecg')`  (ecg was renamed from 'BIO008')

The error runs through: 

`utils.py in verbose(function, *args, **kwargs)
    607         with use_log_level(verbose_level):
    608             return function(*args, **kwargs)
--> 609     return function(*args, **kwargs)`

`ecg.py in create_ecg_epochs(raw, ch_name, event_id, picks, tmin, tmax, l_freq, h_freq, reject, flat, baseline, preload, keep_ecg, verbose)
    303             ecg[None],
    304             create_info(ch_names=['ECG-SYN'],
--> 305                         sfreq=raw.info['sfreq'], ch_types=['ecg']))
    306         ignore = ['ch_names', 'chs', 'nchan', 'bads']`

`/utils.py in verbose(function, *args, **kwargs)
    607         with use_log_level(verbose_level):
    608             return function(*args, **kwargs)
--> 609     return function(*args, **kwargs)`

`array.py in __init__(self, data, info, first_samp, verbose)
     44 
     45         if data.ndim != 2:
---> 46             raise ValueError('Data must be a 2D array of shape (n_channels, '
     47                              'n_samples')`

agramfort, 2016-03-19:
Can you share a file and a script to replicate the pb?

tntdynamight, 2016-03-23:
Here is a file:
https://www.dropbox.com/s/hzry0x2f8d17vbi/ecg_error.fif?dl=0
and the minimal script would be: (using the mne version with the biochannel fix included)

```
import mne

raw_fname='Dropbox/temp/ecg_error.fif'
raw = mne.io.Raw(raw_fname, preload = True) 
raw.filter(1, 45, n_jobs=2)

ecg_epochs = create_ecg_epochs(raw, tmin=-.5, tmax=.5,ch_name='BIO008')
```

closed 2016-03-23 by agramfort