mne-tools/mne-python #3043: BUG: raw.plot_psd() with seeg

kingjr, 2016-03-21:
`raw.plot_psd()`

doesn't has seeg as default.

Eric89GXL, 2016-03-21:
Can you fix it? It would be good to have for 0.12. Perhaps it should use `_pick_data_channels`

kingjr, 2016-03-21:
Yes, as soon as https://github.com/mne-tools/mne-python/pull/3034 is merged. Else it increases the risk of me doing stupid commits between branches.

kingjr, 2016-03-22:
The viz function recently written by  @yousrabk actually surprises me a little: `picks` is actually used as a list of `picks`. 

[IIUC](https://github.com/mne-tools/mne-python/blob/master/mne/viz/raw.py#L411), when `picks` is None, then all channels are regrouped together, as opposed to being regrouped by type? 

I would personally expect `picks` to just be a list of channels, as usual, and the function would then have to find which channels belong to which type.

Furthermore, the regrouping may not be the main use-case IMO. Users are most likely to want to detect bad channels at this stage. And `raw.plot()` plots individual channels.

So I'm personally +1 to change the default behavior before the release. Opinions? @yousrabk?

agramfort, 2016-03-22:
sounds like a bug indeed

Eric89GXL, 2016-05-11:
This one has been fixed, let me know if it doesn't work for you @kingjr

closed 2016-05-11 by Eric89GXL