mne-tools/mne-python #3044: implement robust averaging the SPM way

agramfort, 2016-03-21:
cf mailing list

kingjr, 2016-03-21:
Are you convinced the SPM way would be useful? I have trouble finding in which kind of data distribution it would be sig. better than a mean of 15-85 percentile.

agramfort, 2016-03-21:
no I am not. I agree we should test and compare first

jona-sassenhagen, 2016-03-23:
As by the no free lunch theorem, wouldn't it make sense to offer multiple options?

kingjr, 2016-03-23:
I think @jona-sassenhagen is offering his help :)

jona-sassenhagen, 2016-03-23:
Nope - the next thing I'll be doing is implement the regularization for linear_regression_raw you suggested to me months ago.

kingjr, 2016-03-23:
OK don't hesitate to ping me, this is close to what we're planning on doing
for the gsoc.

MadsJensen, 2016-03-24:
I will be happy to write a comparison over the weekend if that can be a help. Then we can discuss what and how to implement afterwards.

I think added a param to the average call could make sense, with options for percentile in non-SPM robust averaging and weight option for SPM robust average

kingjr, 2016-03-24:
+1, I would pass a 'median' option too, which is the fastest. Thanks @MadsJensen

agramfort, 2016-03-24:
+1 for the benchmark. It would be great if you can show a comparison on our
testing datasets (sample, somato and spm faces)

MadsJensen, 2016-03-28:
I have some progress but I am still working on the SPM robust averaging function. So far there is a mean, median, and robust averaging (kingjr's function) implemented in the analysis. 

I would like to hear if anybody has any thoughts on the best way to quantify the difference between the different methods, I feel that just inspecting plots does not suffice.

[Link to nbviewer](https://nbviewer.jupyter.org/github/MadsJensen/mne_compare_averages/blob/master/Compare_different_approaches_to_averaging.ipynb)
[Link to notebook](https://github.com/MadsJensen/mne_compare_averages/blob/master/Compare_different_approaches_to_averaging.ipynb)

agramfort, 2016-03-28:
thanks heaps

did I miss but did you also include the results obtained with SPM?

to facilitate comparisons it would better to display side by side results
obtained
with the different methods. Can you do it?

thanks again

kingjr, 2016-03-28:
To compare the results, I would make up synthetic data with an imbalanced mixture of data and noise with different SNR), then plot individual (1D e.g. 1 time point, 1 channels) samples together with the estimated mean. e.g.
```
o o o oo oo oo o       o o         o
            |
```

MadsJensen, 2016-03-28:
@agramfort  Sure, i'll do that. No there is no SPM results yet, when I got the a python version of the SPM robust averaging working I will add SPM results. Or did you just think about comparing with the SPM version to the robust, median, mean already available?

@kingjr  thanks for the suggest I will do something like that.

agramfort, 2016-03-29:
>
> @agramfort <https://github.com/agramfort> Sure, i'll do that. No there is
> no SPM results yet, when I got the a python version of the SPM robust
> averaging working I will add SPM results.
>
I would just call the matlab code if it's for checking. If there is a good
reason to add it we'll port it. That's my motivation behind these tests.

> Or did you just think about comparing with the SPM version to the robust,
> median, mean already available?
>

what you call robust is more a trimmed mean. You're not doing the same as
what SPM does when it says "robust“ averaging.

jona-sassenhagen, 2016-04-12:
Can we have something like

```
epochs.average(method=fun)
```

where `fun` can be e.g. `np.mean, np.median`, or something more complex we provide (e.g. `mne.stats._robust_mean`? Defaulting to `np.mean`.

agramfort, 2016-04-12:
sure but why would we want to do this ? I am just provoking but I'd like to
see evidence on data to justify complexifying the API.

jona-sassenhagen, 2016-04-12:
I think *if* we decide on implementing more ways of combining evokeds/TFRs (e.g. robust averaging), API wise, this would be very clean and allow users to easily plug in their own.

Eric89GXL, 2016-04-12:
I'm actually arguing that it's simple enough *not* to need to add to the
API :)

agramfort, 2016-04-12:
>
> I'm actually arguing that it's simple enough *not* to need to add to the
> API :)

I have to admit I agree but it's better if we allow to do it without having
to hack
private attributes etc.

kambysese, 2016-04-12:
I am failing to see any remarkable differences between the averaging approaches  in @MadsJensen notebook, is that a good/bad/moot point with respect to plain old averaging? Is it the case that sample datasets have little variability to make the point? is that what @kingjr implying?

kingjr, 2016-04-13:
@kambysese you need to test it on artefacted data (with blinks etc) where the noise distribution is of a different order of magnitude.

kambysese, 2016-04-13:
Any interest in seeing outcomes on infant auditory data (if you believe in that sort of thing:)?

kingjr, 2016-04-13:
Yes. (And yes)

agramfort, 2018-08-05:
note that robust averaging is averaging with sample weights so it's not any possible reduction function along the epochs axis but internally yes we an work with callables