mne-tools/mne-python #3045: raw.plot - ylim too small to visualize data

srcole, 2016-03-21:
I'm creating a raw object with RawArray, saving it, and then loading it with io.Raw. My data is in units of microvolts, so has values on the order of 10-100. When I try to plot with `raw.plot()`, I have to zoom out dozens of times in order to see my data (i.e. the ylim is set for something <<1).

To test this with the sample data, I modified one of the example data sets (from [this tutorial](http://martinos.org/mne/stable/auto_examples/plot_from_raw_to_epochs_to_evoked.html#sphx-glr-auto-examples-plot-from-raw-to-epochs-to-evoked-py)) to have values on the order of 1, and I again had the issue of needing to zoom out a lot in order to see the data. This is the code to do that:

```
import numpy as np
import mne
from mne import io
from mne.datasets import sample

# Import sample data
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = io.Raw(raw_fname)
raw.plot()

# Change magnitude of the first few channels of sample data
C = 5
T = len(raw.times)
rawnew = np.zeros((C,T))
for c in range(C):
    rawnew[c] = raw[c][0][0] * 10**12
    
info = mne.create_info(ch_names=[str(x) for x in range(C)],ch_types=['eeg']*C,sfreq=150)
raw = mne.io.RawArray(rawnew, info, verbose=False)
raw.save('./foo_raw.fif', overwrite=True)

# Load and plot magnified sample data
raw = io.Raw('./foo_raw.fif')
raw.plot()
```

I don't see an option to change this ylim in order to support visualizing data at an arbitrary magnitude. Not sure if I'm using the toolbox incorrectly, as I've just started trying it.

Eric89GXL, 2016-03-21:
Try the `scalings` parameter of `raw.plot()` to see if it does what you need

http://mne-tools.github.io/stable/generated/mne.io.RawFIF.html#mne.io.RawFIF.plot

srcole, 2016-03-21:
yup. missed that. thanks for the quick response!
Is there a forum for these sorts of questions, or are issues the way to go?

closed 2016-03-21 by srcole

Eric89GXL, 2016-03-21:
The mne_analysis listserv is also a good place to email/look, but an issue is fine (if we didn't have the option then it would indeed be a bug)