mne-tools/mne-python #3047: Infomax does not work on toy example

cbrnr, 2016-03-22:
While discussing https://github.com/scot-dev/scot/issues/160, @mbillingr created a little toy example which shows that standard Infomax fails whereas FastICA works:

```Python
import numpy as np
from mne.preprocessing.infomax_ import infomax
from sklearn.decomposition import FastICA
import matplotlib.pyplot as plt

t = np.arange(1000)
a = np.sin(t * 2*np.pi*0.004)
b = np.sign(np.sin(t * 2*np.pi*0.006))
x = np.vstack([a, b]).T

# optionally mix the signals
# x = np.dot(x, [[1.0, 0.9], [0.9, 1.0]])

u = infomax(x, extended=False)
y = np.dot(x, u)

fi = FastICA()
z = fi.fit_transform(x)

plt.figure()
for i, l, v in zip([1, 2, 3], ['Sources', 'Infomax', 'FastICA'], [x, y, z]):
    ax = plt.subplot(3, 1, i)
    ax.plot(v)
    ax.set_xticks([])
    ax.set_yticks([])
    ax.set_ylabel(l)
    ax.set_ylim([-1.1 * np.max(v), 1.1 * np.max(v)])
plt.show()
```

[image]

Note that only standard Infomax doesn't work - if we set `extended=True`, the decomposition works just fine.

In addition, I ran this example with the [R `ica` package](https://cran.r-project.org/web/packages/ica/index.html) - there, Infomax correctly resolves the sources even in the standard implementation (extended works too).

Could you comment on this issue? Is standard Infomax expected to fail on the toy example data? Or is it a bug? In any case, the MNE implementation of standard Infomax behaves differently than the implementation in the R package.

agramfort, 2016-03-22:
this is a question for @jmontoyam maybe after thursday evening :)

cbrnr, 2016-03-22:
Small update 1: I ran the same toy example with EEGLAB's Infomax implementation. There, standard and extended Infomax are able to recover the sources. However, if I mix the signals (i.e. uncomment the line from the example), both variants are in general not able to recover the original signals.

Update 2: The ICA variants implemented in the R `ica` package also cannot separate a mixture of the two toy signals (but if not mixed, they are able to separate them).

In summary, it seems like it is expected that Infomax (and likely other variants of ICA) do not work on the toy data (but why exactly, what are the assumptions?).

dengemann, 2016-03-22:
> In summary, it seems like it is expected that Infomax (and likely other variants of ICA) do not work on the toy data (but why exactly, what are the assumptions?).

yes we made such observations in the past when proposing to implement infomax in sklearn. Most basic ICA tests would fail. I forgot what was our discussion at that point.

jona-sassenhagen, 2016-03-22:
From a practical standpoint, Is non-extended infomax used much? I doubt so.

If I could change the API at will today, I'd remove the "infomax" method and make fit_params default to `"extended":True`, so it takes work and effort to turn it off.

Not saying it's not important to figure out why infomax is behaving like this.

mbillingr, 2016-03-23:
> and make fit_params default to "extended":True, so it takes work and effort to turn it off.

Actually, the documentation says the default is `True`, although it is implemented as `False`. I guess that should be fixed, one way or the other.

> yes we made such observations in the past when proposing to implement infomax in sklearn. Most basic ICA tests would fail. I forgot what was our discussion at that point.

@dengemann That sounds scary. Have you tried this with other implementations of Infomax as well?

cbrnr, 2016-03-23:
@jona-sassenhagen, yes - probably nobody is and should be using standard Infomax on EEG/MEG data. Extended Infomax should definitely be the default.

Extended Infomax improves upon standard Infomax because it determines for each component if it is super- or sub-gaussian. We should try to implement AMICA (cf #1796), since this variant improves Infomax further by allowing mixtures of Gaussians for each component.

cbrnr, 2016-04-21:
Closing since the numerical issue was addressed in #3049, and it is probably expected that the toy example does not work for Infomax ICA. It's not an MNE issue to find out exactly why Infomax fails on the toy example.

closed 2016-04-21 by cbrnr