mne-tools/mne-python #3052: Q? decimation copy in ICA

kingjr, 2016-03-22:
@dengemann Why do we `copy` data when decimating? https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L367

Eric89GXL, 2016-03-22:
Have you followed `data` in the subsequent calls to see if it gets modified in-place anywhere? Might find the answer that way...

dengemann, 2016-03-22:
> ? Might find the answer that way...

Yes this is the reason.

Eric89GXL, 2016-03-22:
FYI the only other reasons (beyond avoiding in-place modifications) I can think of would be 1) to get data aligned for SIMD operations, but I doubt that's what it's fore, or 2) to save memory if the "new" version is to be stored, but it doesn't look like that's the case either

kingjr, 2016-03-23:
But if it's modified in place, it should be the same whether it's decimated or not, shouldn't it? Or is it because having adjacent columns speeds up the computation?

dengemann, 2016-03-23:
> But if it's modified in place, it should be the same whether it's decimated or not, shouldn't it? Or is it because having adjacent columns speeds up the computation?

thinking about it we could actually avoid it, I guess we did it in the first place assuming that we would update the data but this is actually not the case. It's only used for fitting. Maybe worth testing, we could gain memory performance.

Eric89GXL, 2016-03-23:
> But if it's modified in place, it should be the same whether it's decimated or not, shouldn't it

Oh... right.

> Or is it because having adjacent columns speeds up the computation

This can matter for SIMD instructions via OpenBLAS or MKL but I doubt it's really at play here.

> thinking about it we could actually avoid it ... Maybe worth testing

Agreed. @kingjr if you follow the code and verify that it doesn't modify in place (sounds like it won't) then we should be able to get rid of the copy.

kingjr, 2016-03-23:
I've only followed a part of the code, I'd be reassured if @dengemann validated it.

Eric89GXL, 2016-03-23:
@dengemann do you have time? If not I can take a look and verify

dengemann, 2016-03-23:
@Eric89GXL @kingjr I am pretty sure that it does not get modified, at least not intentionally.

As long as this overwrites a copy: https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L371

the returned data does not have the bad segments. Yes could also be improved with nan-based routine or an annotation based API ...

PR #3059 (2016-03-24): MRG: Remove extra copy
Closes #3052.

@dengemann feel free to merge if you're happy.
closed 2016-03-24 by agramfort