mne-tools/mne-python #3061: BUG: get_brainvision_events does not return neuromag-formatted events

rkmaddox, 2016-03-23:
`mne.io.brainvision.brainvision.RawBrainVision.get_brainvision_events` gives a three-column array where column 0 is index and column 2 is trigger value, as one would expect. However, column 1 stores the duration of the trigger, instead of the standard value of the trigger channel before the change. This creates problems when you try to mask events. So basically, where `raw` is a brainvision Raw object, this:

`mne.find_events(raw)'

gives you a different event that this:

`raw.get_brainvision_events`.

It seems to me that the use of the latter function should be strongly discouraged, or the function deprecated, since its output is incompatible with other functions.

Eric89GXL, 2016-03-23:
@teonbrooks WDYT?

jona-sassenhagen, 2016-03-23:
@teonbrooks maybe we can simply deprecate it? Or is there ever a reason to use it over find_events?

teonbrooks, 2016-03-25:
it should be a hidden function, it's being used internally. we should deprecate the public use and note the particular use for the private function

Eric89GXL, 2016-03-25:
Okay will deprecate, I can make a PR

Eric89GXL, 2016-03-25:
@teonbrooks is `set_brainvision_events` meant to be exposed? I'm not sure what the purpose is.

teonbrooks, 2016-03-25:
I don't know either. I didn't write this function. it was factored out of the code I wrote. closing this for #3074

closed 2016-03-25 by teonbrooks