mne-tools/mne-python #3062: ENH: better solver for rERP, encoder models redux

jona-sassenhagen, 2016-03-23:
Ridge and OLS as implemented in sklearn are much faster than the current solver, for some reason. (The reason is that obviously, something I hack together will always be worse than sklearn.)
They also are better at optionally dealing with sparse input.
I'd like to add them, and make them the go-to with the current one as a fallback.

(This is another incremental step of expanding rERP/linear_regression_raw into an encoder model.)

I'll go about implementing the first one soon.

Q: what's the most economic way of checking if a scipy coc sparse array is truly sparse, or alternatively, what's a good way of checking if it should be treated as sparse by the solver, or converted to dense and solved by a normal solver??

Eric89GXL, 2016-03-23:
> Q: what's the most economic way of checking if a scipy coc sparse array is truly sparse

Depends on what you mean by that, but this would get you some of the way maybe

http://docs.scipy.org/doc/scipy-0.14.0/reference/generated/scipy.sparse.csr_matrix.nnz.html

jona-sassenhagen, 2016-03-23:
E.g. you can currently use this for the solver param and it works better:

`def solver(X, y):
    from sklearn.linear_model import Ridge
    return Ridge().fit(X, y.T).coef_`

Also works with RidgeCV. Not really with Lasso though ..

jona-sassenhagen, 2016-03-23:
Thanks Eric, that's probably it. Unless anybody has a rule of thumb regarding up to what sparsity/density it makes sense to leave X as sparse/use a sparse solver.

Eric89GXL, 2016-03-23:
I think you should always leave it sparse and leave it up to the user to do the right thing in terms of converting it or not

jona-sassenhagen, 2016-03-23:
Sadly, this is not possible for the user-facing rERP function, because exists only inside the function. The user doesn't supply X, they supply the information to construct X, and gets the evokeds.

Eric89GXL, 2016-03-23:
so the rERP function decides whether to construct a sparse matrix or dense array?

jona-sassenhagen, 2016-03-23:
Due to the way we've settled on handling it internally, it always starts out as sparse (see https://github.com/mne-tools/mne-python/blob/master/mne/stats/regression.py#L363), and right now, the solver always converts it to dense (see https://github.com/mne-tools/mne-python/blob/master/mne/stats/regression.py#L254).

jona-sassenhagen, 2016-03-23:
The part where Denis and I settled on using sparse.dia_matrix is handled differently by Chris' approach, by XDawn, by the original implementation, and was handled differently by the initial version of linear_regression_raw. The current version is the only one of these that's sparse, and that was because we saw major performance improvements from that.

Eric89GXL, 2016-03-23:
> Unless anybody has a rule of thumb regarding up to what sparsity/density it makes sense to leave X as sparse/use a sparse solver.

Ahh, that's what you mean by this. I don't know any rule of thumb, but you could do some benchmarking to try to see when approximately it switches.

> The current version is the only one of these that's sparse, and that was because we saw major performance improvements from that.

To start, though, I'd just try to get it working and leave such optimization for later (with notes in the code to come back to it). Trying to optimize with such branching while also expanding / unifying funtionality can make things more difficult. To do it properly I'd verify that the dense and sparse paths give equivalent results (meaning unit tests) and then do benchmarking, etc.

jona-sassenhagen, 2016-03-23:
Actually, I just realized the user can indeed decide on converting to dense by plugging in a solver that converts to dense vs. one that doesn't. But I always like an 'auto' option ... I'll play around with it a bit.

jona-sassenhagen, 2016-03-23:
E.g. the user can do

```
def solver_sparse(X, y):
    from sklearn.linear_model import Ridge
    return Ridge().fit(X, y.T).coef_
```

and

```
def solver_dense(X, y):
    from sklearn.linear_model import Ridge
    return Ridge().fit(X.toarray(), y.T).coef_
```

and then

```
evokeds = linear_regression_raw(raw, events, event_id, solver=solver_dense, ...)
```

closed 2016-03-30 by Eric89GXL

jona-sassenhagen, 2016-03-30:
Nathaniel Smith uses a very memory efficient incremental solver where Xt * X and Xt * y are constructed incrementally. I've finally basically figured out how it works (I think ...), and implemented a special solver for it. It's not necessarily faster, and it is only an asymptotically approximate solution as far as I can tell, but for certain cases, it should be much faster, and in general, it should allow to solve even massive problems in little RAM. (I usually run this on a machine with 250 GB RAM, but I am being told some use their laptops ...)
 
Example results (EEGLAB sample data) looks like this.

Incremental solver:
[image]

Current solver:
[image]

Difference (notice the unit):
[image]

Implementation:
```
def incremental_solver(X, data):
    stride_ratio = 2
    n_times, n_coefs = X.shape
    one_stride = int(n_coefs * stride_ratio)
    loops = (n_times // one_stride) - 1

    XtX_all = np.zeros((n_coefs, n_coefs))
    XtY_all = np.zeros((X.T * data.T).shape)
    for ii in range(0, n_times - one_stride + 1, one_stride):
        reach = (one_stride + (n_times % one_stride)
                 if ii + (one_stride * stride_ratio) > n_times
                 else one_stride)
        XtX_all += (X[ii:ii + reach, :].T * X[ii:ii + reach, :])
        XtY_all += (X[ii:ii + reach, :].T * data[:, ii:ii + reach].T)

    return linalg.solve(XtX_all, XtY_all)
```

I'd like to offer this as an alternative.

PR #3080 (2016-03-30): MRG: Add cholesky solver for rERP
closes #3062