mne-tools/mne-python #3063: topo interactivity broken on evoked.plot()

jasmainak, 2016-03-23:
If I do this:
[image]

i.e.,

```py
epochs.average().pick_channels(['MEG 1143']).plot()
```

I get this:

```py
n [15]: Traceback (most recent call last):
  File "/tsi/doctorants/mainak/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_qt5.py", line 291, in mouseReleaseEvent
    FigureCanvasBase.button_release_event(self, x, y, button)
  File "/tsi/doctorants/mainak/anaconda/lib/python2.7/site-packages/matplotlib/backend_bases.py", line 1891, in button_release_event
    self.callbacks.process(s, event)
  File "/tsi/doctorants/mainak/anaconda/lib/python2.7/site-packages/matplotlib/cbook.py", line 533, in process
    proxy(*args, **kwargs)
  File "/tsi/doctorants/mainak/anaconda/lib/python2.7/site-packages/matplotlib/cbook.py", line 408, in __call__
    return mtd(*args, **kwargs)
  File "/tsi/doctorants/mainak/anaconda/lib/python2.7/site-packages/matplotlib/widgets.py", line 1250, in release
    self.onselect(vmin, vmax)
  File "/tsi/doctorants/mainak/mne-python/mne/viz/evoked.py", line 90, in _butterfly_onselect
    layout=None)
  File "/tsi/doctorants/mainak/mne-python/mne/viz/topomap.py", line 49, in _prepare_topo_plot
    picks, pos = _pair_grad_sensors(info, layout)
  File "/tsi/doctorants/mainak/mne-python/mne/channels/layout.py", line 689, in _pair_grad_sensors
    raise ValueError("No 'grad' channel pairs found.")
ValueError: No 'grad' channel pairs found.
```

also I find the key mappings a bit weird because it interferes with magnifying the plot ...

jasmainak, 2016-03-23:
cc @jaeilepp @jona-sassenhagen

jona-sassenhagen, 2016-03-23:
I don't know the interactive functions at all, but is it possibly related to #3003 ?

jaeilepp, 2016-03-24:
The problem only occurs if you have grad data where none of the channels have a pair, in which case the error seems reasonable. What I suggest instead is to add the number of channels or pairs plotted on the topomap title, like it is here in ``evoked.plot``.

jasmainak, 2016-03-24:
I still don't think it is okay to have errors in interactive mode ...

jaeilepp, 2016-03-25:
But you're trying to do something that's not possible. Another option would be to throw a warning, but to me, error seems more appropriate.

Eric89GXL, 2016-05-11:
We could disallow interactivity altogether, but then people will wonder why it's there. So to me the only options are warning and error (although maybe we should make a more explicit message in this interactive case, about how it's not possible to select), and I lean toward error. @jasmainak WDYT?

jasmainak, 2016-05-11:
okay with error, but maybe we need a better message. Also, it might be better to put it in the text (instead of `loading ...`, say `need corresponding gradiometer channel for interactivity` in red font) than in command line?

Eric89GXL, 2016-05-11:
Ahh right, yes we could also modify the plot, that would make it very clear.

closed 2016-06-29 by Eric89GXL