mne-tools/mne-python #3068: Extend tfr to continuous data

jona-sassenhagen, 2016-03-24:
From what I can tell, it would be fairly straight-forward to extend the TFR functions to continuous data. For example, this seemingly works:

```
fake_evoked = mne.EvokedArray(raw._data, raw.info, tmin=0)
out = mne.time_frequency.tfr_morlet(ev, freqs, 4, return_itc=False)
```

Can we add raw data input for tfr_morlet and the like?

jasmainak, 2016-03-24:
Do you have an estimate of how long this computation will take on sample data?

jona-sassenhagen, 2016-03-26:
On real data, it's fairly long, and the results are huge. On the sample data, EEG channels only, it's under a minute on a reasonable computer. But it's slow, like all TF stuff (unless you have CUDA).

kingjr, 2016-03-26:
We have to decide whether we should loop over time windows , channels or
freqs. The memory load can rapidly become a nightmare.

I don't have my computer with me ATM, but I wrote a gist for computing tfr
on raw with a load param.

agramfort, 2017-07-22:
I think it's better to do it via Epochs with overlaps. You'll be able to remove artifacts etc. closing

closed 2017-07-22 by agramfort