mne-tools/mne-python #3069: Website

teonbrooks, 2016-03-25:
- [ ] the development links says this is v0.11 but we are in dev0.12

- [ ] when hovering over header links, it displays the paragraph symbol, this doesn't seem like the correct behavior.

- [ ] the example figures look funky here:
http://martinos.org/mne/dev/auto_examples/visualization/plot_meg_eeg_fields_3d.html#sphx-glr-auto-examples-visualization-plot-meg-eeg-fields-3d-py

agramfort, 2016-03-25:
can you refresh your browser cash?

Eric89GXL, 2016-03-25:
> when hovering over header links, it displays the paragraph symbol, this doesn't seem like the correct behavior.

That's a symbol that means you can use it as a link and I think it's useful. I've seen it on other sites, too.

teonbrooks, 2016-03-25:
the website is up-to-date for me now. everything looks great!

> That's a symbol that means you can use it as a link and I think it's useful. I've seen it on other sites, too.

ah, ok. sounds fine. 

closing

closed 2016-03-25 by teonbrooks