mne-tools/mne-python #3070: Return self for in-place operations; Enable chaining for in-place operations

teonbrooks, 2016-03-25:
We have some inconsistencies about when we return `self` for in-place operations and when we don't. I think we apply this rule consistently when there is a `copy` param but it seems like we should have it elsewhere. it's useful for chaining. this was recently brought up on the listserv when someone tried to assign the `raw.filter()` output to a variable. If you wanted to `raw.filter().plot()` it wouldn't work.

jona-sassenhagen, 2016-03-25:
I heartily support this.

Note it can probably be made to work without disruptive changes (unless somebody really needs their `raw.filter()` to return `None`...).

One big PR or slower rollout?

kingjr, 2016-03-25:
+1 Some methods already return self, e.g. `pick_types` but it's not consistent.

Eric89GXL, 2016-03-25:
I'd say get as many as you can in one PR, they should be almost exclusively one-liners plus docstring updates

Eric89GXL, 2016-03-25:
(IMO we don't need to test the return value actually exists, so don't need to modify tests -- we'll end up naturally modifying them in the future probably to use the functionality)

Eric89GXL, 2016-03-25:
@teonbrooks @jona-sassenhagen @kingjr one of you have time to take a stab at it in the next week or two? If not I can try. Would be good to have for 0.12.

kingjr, 2016-03-25:
If I have time, I was thinking to devote it to the binder examples.

Eric89GXL, 2016-03-25:
+1 that's a higher priority in my mind

mmagnuski, 2016-03-26:
(this got closed by accident :hamster: )

Eric89GXL, 2016-03-27:
I think @agramfort and @jona-sassenhagen's PRs hopefully take care of most of the cases

teonbrooks, 2016-03-27:
@mmagnuski you're right, there was a misdirected ref that closed it.

jona-sassenhagen, 2016-03-27:
What else needs a copy param? Epochs.savgol?

Eric89GXL, 2016-03-27:
Added `savgol` to my PR

PR #3074 (2016-03-25): MRG: Deprecate get_brainvision_events
Ready for review/merge @teonbrooks.

Closes #3070.
PR #3079 (2016-03-29): MRG: Minor fixes to docstring and more copy params
Lots of minor docstring fixes.

Ready for review/merge from my end.

Closes #3070.
closed 2016-03-29 by Eric89GXL