mne-tools/mne-python #3073: ENH: `picks` in raw plot #easy

kingjr, 2016-03-25:
`raw.plot` doesn't have a `picks` param. IMO it should be added.

Eric89GXL, 2016-03-25:
To conserve memory this can't just use `raw.pick_channels`, so I suspect it won't be a trivial change. A lot of drawn object counts and such will need to change since we won't be able to rely on e.g. info['nchan'] anymore. But if someone is motivated it could be helpful.

kingjr, 2016-03-25:
hadn't realized, I removed `easy` from the title.

Note that this particularly useful for seeg, where a lot of channels can be bad, and take unnecessary room.

agramfort, 2016-03-25:
we could also have a way to restrict views to canonical selection like
mne_browse_raw does it.

dengemann, 2016-03-25:
> we could also have a way to restrict views to canonical selection like
mne_browse_raw does it.

sounds good.

jaeilepp, 2016-03-31:
Could we actually do some inferring of channel sets from the channel locations in the data? This way we wouldn't have to have pre-set channel sets for each data format. I suppose it wouldn't be terrible if the inferring doesn't work accurately in all the cases as this is just for making channel sets for plotting time series.

agramfort, 2016-03-31:
Just look how it is done in plot_evoked_topomap

Eric89GXL, 2016-05-11:
> Could we actually do some inferring of channel sets from the channel locations in the data? This way we wouldn't have to have pre-set channel sets for each data format. I suppose it wouldn't be [...]

I think ideally we would offer both modes. I suspect Neuromag folks are already surprised we don't have a way of showing data in the standard channel groups, especially since we already have the selections available.

Right now we plot channels by type, but we don't have to do it that way. So maybe we could:

1. Add `order` kwarg to `raw.plot` (and maybe also `epochs.plot` since it does similar things).
2. If array, plots channels (and only those channels) in the given order.
3. If `'type'` (default), plot by channel type like we do now.
4. If `'position'`, it tries to go by position through the channels in some sensible way.
5. If `'selection'`, it plots with channel type selections (might only work for Neuromag data?).

Come to think of it, I don't actually know how mne_browse_raw does it -- does it iterate over selections?

In any case, I think that would cover all the use cases we've talked about. Under the hood it would mostly I think be some "fun" indexing problems to sort out.

agramfort, 2016-05-11:
I like the order API you suggest

closed 2016-06-09 by agramfort