mne-tools/mne-python #3075: FIX: tutorials title layout

kingjr, 2016-03-25:
Several sections of the tutorials have their title as first bullet:

http://martinos.org/mne/dev/manual/index.html

e.g.
```
Cookbook
A quick run-through of the basic steps involved in M/EEG source analysis.

    * Cookbook
```

I guess the advantage is that the bullet has an external link, but it seems redundant.

agramfort, 2016-03-25:
PR welcome :)

Eric89GXL, 2016-05-11:
I've noticed this before and poured over the sphinx docs trying to find a way to exclude items from a TOC, or to start the TOC from e.g. one level deep without success. So I'm not optimistic about solving this one, but I'd love to be proved wrong.

Eric89GXL, 2016-07-18:
I opened an issue on Sphinx-doc (https://github.com/sphinx-doc/sphinx/issues/2790) with an offer to make a PR, we'll see if it gets any traction

Eric89GXL, 2017-02-14:
I looked into it and this is a pain at the Sphinx level. So I'll close this as won't fix

closed 2017-02-14 by Eric89GXL