mne-tools/mne-python #3078: Much too hard to plot an array as a topomap?

jona-sassenhagen, 2016-03-26:
There isn't a simple one-liner to plot an array the shape of your sensors as a topomap right? Something like:

`mne.viz._plot_topomap_for_dummies(array, info)`

(Assuming `info` has as many data channels as array has values.)

You always have to mess with layouts etc. This is easier in EEGLAB, which can plot based on just an array and chanlocs.

IME it makes quite a bit of code much harder than need be, both internally and for users.

Or am I missing a function/usage?

agramfort, 2016-03-26:
I'd be fine with a new

plot_array_topomap

as I often also end up using an EvokedArray just for this and times are not
always meaningful

wmvanvliet, 2016-03-27:
We don't necessarily need a new function. The difficulty mostly stems from `plot_topomap` not being able to take coordinates in 3D. It requires coordinates already mapped to 2D (actually, layout might not even properly do this, as there is an offset of box_width/2 and box_height/2). The best method is to use the private function `_find_topomap_coords`. We could make that function public. Even more convenient would be to make `plot_topomap` detect when coordinates are given in 3D (`if coords.ndim == 3`) and call the `_find_topomap_coords` function.

wmvanvliet, 2016-03-27:
@dengemann, care to weigh in?

jona-sassenhagen, 2016-03-27:
I'd still like to see a function that works off of info and an array.

dengemann, 2016-03-27:
Sorry having difficulties to connect. Isn't auto layout taking care of everything? Not sure we need a new API. I think we can still improve our default behavior, I have the suspicion that @jona-sassenhagen is saying that something is not user friendly ;)

dengemann, 2016-03-27:
> as I often also end up using an EvokedArray just for this and times are not
always meaningful

Ah I see. Yes me too. @jona-sassenhagen is this your point?

jona-sassenhagen, 2016-03-27:
I just think it should be possible to plot an array without any boilerplate, such as extra function calls, mock objects etc. All the information is in info, after all.

dengemann, 2016-03-27:
Ok agreed.

agramfort, 2016-03-27:
please share a code snippet of how you would like to write stuff.

kingjr, 2016-04-13:
I'm also +1 on this.

jona-sassenhagen, 2016-04-19:
>please share a code snippet of how you would like to write stuff.

You mean API?

The easiest solution would be to allow mne.viz.plot_topomap to take an info instance (instead of only a pos array). Then, instead of doing

```
    picks, pos, merge_grads, _, ch_type = _prepare_topo_plot(evoked,
                                                             ch_type,
                                                             layout=None)
    plot_topomap(arr, pos)
```

which IMO is not very beginner friendly and requires accessing private functions, one could do

`plot_topomap(arr, info)`

jona-sassenhagen, 2016-04-23:
@agramfort what do you think?

agramfort, 2016-04-23:
I don't know the details but I like a

plot_array_topomap(arr, info)

just make it work without breaking the API and duplicating code :)

jona-sassenhagen, 2016-04-23:
Adding a param for plot_topomap is cleaner than a new function, don't you think?

agramfort, 2016-04-23:
I would say no if you make params redundant or conflicting (one param
shadows another)

dengemann, 2016-04-23:
+1 for new function!

jona-sassenhagen, 2016-04-23:
Hm ... plot_array_topomap doesn't seem like a good name because plot_topomap already plots an array as a topomap. In fact, the two functions should be 99% identical, the only difference being that plot_topomap has pos as its second argument, and plot_array_topomap an info.

In fact, as @wmvanvliet points out, it would be roughly equivalent to

plot_topomap(array, _find_topomap_coords(info))

How about overloading the pos keyword for plot_topomap so it also allows an info object?

@wmvanvliet s suggestion of making _find_topomap would partially ameliorate the problem, but we probably also have to deal with merging grads.

Okay, this is my last objection, if you're still not convinced, I'm starting work on plot_array_topomap :)

agramfort, 2016-04-23:
On Sat, Apr 23, 2016 at 5:24 PM, jona-sassenhagen <notifications@github.com>
wrote:

> In fact, as @wmvanvliet <https://github.com/wmvanvliet> points out, it
> would be roughly equivalent to
>
> plot_topomap(array, _find_topomap_coords(info))
>
>
if this really works I am fine with overloading the pos param

Eric89GXL, 2016-04-23:
It sounds like it probably is easy enough to do `if isinstance(pos, Info): ...`, so +1 on that for me. Then it's just a few lines of code to extract the relevant positions (probably de-duplicated with where it's already done elsewhere) and a docstring update.

@jona-sassenhagen if you can do it in the next couple of days it can go in 0.12

jona-sassenhagen, 2016-04-23:
I will try, please assign me Eric :) But make it .12, .13.

closed 2016-04-25 by jaeilepp

mdnunez, 2016-05-12:
There was not an example of new usage here. If a user has an arbitrary data array but a known EEG montage (e.g. an EGI 128 electrode cap) is the correct usage then the following?

```
import numpy as np
import mne
from mne.viz import plot_topomap
tempweights = np.random.rand(128) # 128 channels of arbitrary EEG data
egimontage = mne.channels.read_montage('GSN-HydroCel-128')
tempinfo = mne.create_info(egimontage.ch_names[3:], 1000, ch_types='eeg', montage=egimontage)
plot_topomap(tempweights, tempinfo)
```

jona-sassenhagen, 2016-05-12:
I guess that works, although my idea was to use this when you have existing MNE objects and do some calculation on them (e.g., channel activity correlated with RT within-subjects), and just re-use the info from the objects. In that case, you usually have to pick only data channels.

I'll do an example, unless you want to do one.

mdnunez, 2016-05-13:
If a few of the functions can stand alone, mne may be more attractive to those users who want to use their own or other modules. Thanks, in the future I will contribute but I want to learn a little more about mne first.

jaeilepp, 2016-05-13:
There is an example usage of this in http://mne-tools.github.io/dev/auto_examples/decoding/plot_decoding_csp_space.html#sphx-glr-auto-examples-decoding-plot-decoding-csp-space-py, but I think it should be added to vis tutorials too.

PR #3165 (2016-04-25): MRG plot_topomap from info
Closes #3078

Try:
```
import mne
data_path = mne.datasets.sample.data_path()
evoked_fname = data_path + '/MEG/sample/sample_audvis-ave.fif'
for type in ('mag', 'grad'):
    evoked = mne.read_evokeds(evoked_fname, condition=1).pick_types(eeg=False, meg=type).detrend()
    arr = evoked.data[:, 171]
    mne.viz.plot_topomap(arr, evoked.info);
```