mne-tools/mne-python #3082: Possible areas of improvements

teonbrooks, 2016-03-28:
I've been talking to a few colleagues about their experience with MNE, some of them coming from other packages, here are some of their experiences:

- [ ] Figuring out how to grand average source data: there seems to be a function for evoked and TF data but not for source. I ended up morphing them creating a multidimensional matrix np.array and doing the average myself. But couldn't find any documentation about this.

- [ ] Took me awhile to figure out that I needed to baseline my epochs before computing noise covariance. Although I did end up finding that answer in the manual.

-- could be address here in the manual: http://mne-tools.github.io/dev/manual/source_localization/covariance.html

- [ ] Another thing that could use more elaboration is the use of the lambda2 parameter in apply_inverse. Why is it set to 1/2**3 most of the time? In fieldtrip, the lambda parameter is adjusted essentially to whatever gives you the best data... which is confusing and not ideal so curious as to how this value was settled on.

- [ ] A better tutorial for how to deal with make_bem_watershed. And what to do with the error that one surface intersects with another. This is sort of a freesurfer problem... but since it is an MNE function it would be really useful to have some documentation on how to deal with this.

Eric89GXL, 2016-03-29:
> how to grand average source data

```
stc_ave = stc_1 + stc_2 + ...
```

> Took me awhile to figure out that I needed to baseline my epochs before computing noise covariance

You got warnings when you tried to compute the covariance without baselining, though, right? You should have. If you didn't, that's a bug.

> Another thing that could use more elaboration is the use of the lambda2 parameter in apply_inverse. Why is it set to 1/2**3 most of the time

It seems to give reasonable values in most situations. We could potentially automate it (we do have some SNR calculations available) but I guess nobody has felt the need to do it yet. I'm not sure how much we would gain from optimizing it.

> A better tutorial for how to deal with make_bem_watershed. And what to do with the error that one surface intersects with another.

I assume getting a better MRI is not a better option? :) I've never had to deal with it so can't help much there.

Do you want to start a PR for the docs and we can iterate?

dengemann, 2016-03-29:
> Took me awhile to figure out that I needed to baseline my epochs before computing noise covariance

more precisely it should be zero-mean, which is typically also achieved by filtering. I think we already say that here and there.

teonbrooks, 2016-03-29:
cc: @kdoelling1919

> It seems to give reasonable values in most situations. We could potentially automate it (we do have some SNR calculations available) but I guess nobody has felt the need to do it yet. I'm not sure how much we would gain from optimizing it.

is there any documentation or somewhere to point to where this chosen value comes from?

teonbrooks, 2016-03-29:
> more precisely it should be zero-mean, which is typically also achieved by filtering. I think we already say that here and there.

yeah, it's in the docstring, we can eventually just populate the manual page with this info.

I will eventually get around to this but I can't get to this yet. just adding to my list...totally open to anyone who wants to contribute :)

choldgraf, 2016-04-01:
I don't know anything about any of the methods you guys are talking about :)

BUT, I am +1000 on beefing up the tutorials, manuals, etc. If you guys ever need someone to look over material with fresh (maybe newbie) eyes, then let me know.

agramfort, 2016-04-01:
week ends are good to write docs :)

Eric89GXL, 2016-04-01:
@choldgraf yes that would help. If you still remember what was difficult to do you could make a of what was difficult to understand how to do in mne-python, and we can improve our documentation of it -- you can make an issue or share a google doc or something. And if you do read through the docs and see misleading things, add them there or make a PR. My eyes are a bit too stale to see these things :)

choldgraf, 2016-04-01:
sounds good - yeah my plan is to crank out some docs. I've got a few things on the to-do list to work through before I get to it, but it will happen :)

agramfort, 2017-07-22:
can we close this now @teonbrooks ?

larsoner, 2018-04-04:
Closing due to non-response, feel free to reopen if not addressed

closed 2018-04-04 by larsoner