mne-tools/mne-python #3083: Add two methods Epochs, Evoked

teonbrooks, 2016-03-29:
It would be super useful to have these methods added to the Epochs, and Evoked.

`raw.time_as_index` and `raw.index_as_time`

kingjr, 2016-03-30:
What s the use case ? Can you just divide by `info['sfreq']`?

teonbrooks, 2016-03-30:
yes, but it is a function that exists. it seems like it should be propagated since it's convenient, it takes into account the rounding business as well

teonbrooks, 2016-03-30:
it's mostly needed for time_as_index. it deals with the fact that you might have a baseline period and just by multiply by the sfreq isn't enough, you would have to adjust for baseline

agramfort, 2016-03-30:
time_as_index makes sense. The other one is just a matter of e.times[idx]

Eric89GXL, 2016-04-14:
@agramfort okay with you to deprecate `raw.index_as_time` then and point people toward `inst.times[idx]`?

agramfort, 2016-04-14:
yes

PR #3132 (2016-04-18): MRG: Mixin time_as_index
Closes #3083.

Ready for review/merge from my end.
closed 2016-04-18 by agramfort