mne-tools/mne-python #3086: Handling of stimulus categories

jjnurminen, 2016-03-31:
I'm writing code that needs to handle multiple stimulus categories. I noticed couple of things:

mne.Epochs.get_data() concatenates epochs from all categories, i.e. returns a matrix of (n_total_epochs x n_channels x n_samples). However, the order of concatenation is not clear; it does not go according to Epochs.event_id. I would propose either documenting the order of concatenation, or alternatively returning a set (list? dict?) of arrays corresponding to different categories (however the latter option would break existing code).

Another issue is that it does not seem possible to create Evoked objects with multiple categories from arrays: mne.EvokedArray method only takes a single category. One solution would be to make every category into a separate Evoked object, but my goal is to save several categories into one evoked fiff file. Is there any way to do this?

agramfort, 2016-03-31:
are you aware of the hierarchical event_id ?

https://mne-tools.github.io/dev/auto_examples/plot_from_raw_to_epochs_to_evoked.html#sphx-glr-auto-examples-plot-from-raw-to-epochs-to-evoked-py

basically if

event_id = {'Auditory/Left': 1, 'Auditory/Right': 2}

then you can just do

epochs['Auditory']

and you will get all auditory conditions but can also do

epochs['Left']

and you will get all left conditions.

regarding the order of get_data it's the order coming from the file.
So indeed all conditions are mixed up if you're paradigm was like that

jjnurminen, 2016-03-31:
@agramfort Yep, I'm aware of that. I'm leaning towards handling one category at a time. Starting from these, is there a good way to create an evoked object with several categories?

So basically I would like to do 
load epochs of category1 -> do some processing -> average into evoked category1
load epochs of category2 -> do some processing -> average into evoked category2

and then save both evoked categories into the same fiff file.

agramfort, 2016-03-31:
something like this?

mne.concatenate_epochs([epochs['Left'], epochs['Right']])

then you can always call the average method to average all the selected
epochs

jjnurminen, 2016-04-01:
@agramfort If I use mne.concatenate_epochs, I get one Epochs object with all categories included. However if I then call Epochs.average(), it returns an Evoked object which contains an average of ALL epochs (even though they correspond to different categories).

To me it would be more logical if Epochs.average() did the averaging separately for the different categories and returned an Evoked object with several categories. Or am I missing something?

agramfort, 2016-04-01:
> @agramfort <https://github.com/agramfort> If I use
> mne.concatenate_epochs, I get one Epochs object with all categories
> included. However if I then call Epochs.average(), it returns an Evoked
> object [...]

the logic is that average will indeed average ALL. If you want to average a
subset of epochs just select them eg

epochs['Auditory']
epochs[[1, 3, 4]]
etc.

and then call average()

does it make sense?

jjnurminen, 2016-04-01:
@agramfort Well it does, but then how do I save all averaged categories into a single fiff file?
For that, I would need a single Evoked object that has all the categories, right?

agramfort, 2016-04-01:
>
> @agramfort <https://github.com/agramfort> Well it does, but then how do I
> save all averaged categories into a single fiff file?
> For that, I would need a single Evoked object that has all the categories,
> right?
>
yes you get one evoked per category and then you do

mne.write_evokeds(fname,  [ev1, ev2])

to put ev1 and ev2 in the same file.

http://martinos.org/mne/dev/generated/mne.write_evokeds.html

ok?

jjnurminen, 2016-04-01:
@agramfort Ok! That should do it. I was looking at class methods only and missed write_evokeds.

I think it would be great for Evokeds.average() to do per-category averaging at least as an option, since that's the most common use case. It would return an Evoked object with multiple categories.

Anyway, thanks for the help, I'm closing this now.

jona-sassenhagen, 2016-04-01:
@jjnurminen

```
evokeds = {cond:epochs[cond].average() for cond in conds}
```
I do something like this a lot.

Also linear_regression_raw returns a dict of evokeds.

Eric89GXL, 2016-04-01:
I know @drammock happened to hit this same problem yesterday, so we should try to improve it. @jjnurminen would clearer docstring in `.average` and `.save` have saved you some trouble? (Did you look at the docstring / parameter lists for those, but not see how to accomplish what you wanted?)

drammock, 2016-04-01:
:+1: for the idea of  `.average(by_condition=True)` as a parameterization (should default to `False` for backward-compatability).  In my case I stumbled on the same dict-comprehension solution @jona-sassenhagen mentioned, and if this change happens I think a dict is the most sensible return-type.

Eric89GXL, 2016-04-01:
@agramfort WDYT about adding a parameter? We clearly need to beef up the docstring, but it sounds like at least 3 people would have been abel ot use `by_condition` or some similar param.

agramfort, 2016-04-01:
I think we can fix this with a better docstring

evokeds = [epochs[condition].average() for condition in event_id]

is really that hard to write/find?

jona-sassenhagen, 2016-04-01:
I for one am happy with the dict comprehension idiom.

Also it would be simpler if the average method would always return evokeds IMO, not suddenly a list.

jona-sassenhagen, 2016-04-01:
So I think docstring/example updates should be preferred.

Eric89GXL, 2016-04-01:
I'll reopen and edit the title/description to reflect the consensus

jjnurminen, 2016-04-03:
Thanks for weighing in @Eric89GXL . I think it's a good idea to improve the docstring, at least for average(). 

FWIW, I find it a bit counterintuitive that Epochs is multi-category, but Evoked is not. It would also be good to document this fact more clearly. (IMO, the ideal solution would be to make Evoked support multiple categories, implement __getitem__ etc. After all, there is not much logical difference between Evoked and Epochs. Then one could just average() an Epochs and get Evoked with the same categories.)

teonbrooks, 2016-04-03:
I'm guessing that we can't have all the Evokeds within the same object similar to how Epochs stores conditions due to compatibility issues? indeed I agree with @jona-sassenhagen about the utility in dictionaries to refer to conditions since they are in one object that you can reference. it's harder to deal with evokeds when they are in a list because it takes more steps to refer to a particular evoked (you end up needing to search the comment of each evoked for the condition name when it is in a list which is pretty annoying)

jona-sassenhagen, 2016-04-03:
@jjnurminen I'm not sure this can be achieved without a total reworking of the Evoked class, which will not happen. Maybe the same can be achieved with a new kwarg and a revamped example here and there.

Eric89GXL, 2016-04-03:
I agree that if we were starting from scratch it would be better to make it
act more like Epochs. But now we (and our users) have a lot of code that
depends on the current pattern. Since it's a fundamental/core object,
making such changes is thus difficult.

We could think about extending Evoked to handle multiple conditions in a
way that's backward compatible, but I suspect it would require a lot of
thinking, time, and revisions to get right. I wouldn't stop sometime from
trying if they were motivated to try it, but there's no guarantee in my
mind that the work would pay off as something that ends up being mergeable.

teonbrooks, 2016-04-04:
`mne.Evokeds` ;)

agramfort, 2016-04-04:
The reason for my original design choice:

- if you have event_id = {'Auditory/left': 1, 'Auditory/right': ",
'Visual/left': 3, 'Visual/Right':4}, what is a condition? is it the visual?
the visual left etc? then if you have an Evoked that contains all
conditions and you do evoked.plot() does it plot all available conditions?
- If you do evoked.data to get the data how do you know what condition it
corresponds to?

I am just saying that there would be API consequences in doing what you
suggest and I mean more code complexity. The Evoked class is already quite
complex and the current situation is to me flexible enough to
allow pretty much anything (no?).

To have a dict of Evoked just do a dict comprehension it is one line of
code and if you want
to save all this to one file it's also one line of code.

my 2c

jjnurminen, 2016-04-04:
Sure @agramfort, these are just design decisions and one can make arguments either way. I'm very much aware that it wouldn't be a simple change to to make.

Just let me know if you can use any help with the docs.

agramfort, 2016-04-04:
>
> Just let me know if you can use any help with the docs.
>
it's always more than welcome. Make sure you can build it on your machine
and then it's a matter of editing the .rst files or just the function
docstrings.

jjnurminen, 2016-04-09:
Quick question: if I'm just doing some minor doc fixes, is it feasible to create PRs in the GitHub web interface? Or should one always set up a full dev environment and work via that?

agramfort, 2016-04-09:
you can just click the edit button on github and it will make PRs
automagically

jjnurminen, 2016-04-09:
OK. I was thinking that maybe one always needs to run tests in the devenv before creating the PR, or something.

agramfort, 2016-04-09:
start slowly. Yes doing a proper  and running all tests before pushing is
the best practice.

jjnurminen, 2016-04-10:
OK, I set up everything according to the contributing doc and ran 'make', which runs the nosetests.
I left it running for 20 hours and it still hadn't completed (or at least didn't return to command prompt). It printed out several lines like:

[Parallel(n_jobs=2)]: Done   2 out of   2 | elapsed:    0.0s finished

After 20 hours I pressed Ctrl-C and got a traceback followed by:

Ran 40 tests in 82660.137s
OK (SKIP=12)

So I'm a bit confused here. Did it complete or not? Is my old PC not up to task (4GiB of RAM and dual core Athlon CPU?)

agramfort, 2016-04-10:
> OK, I set up everything according to the contributing doc and ran 'make', which runs the nosetests.
> I left it running for 20 hours and it still hadn't completed (or at least didn't return to command [...]

that's really long and a bit weird. The full test suite should run in
30 mins max. See for example what you should see:

https://travis-ci.org/mne-tools/mne-python
https://travis-ci.org/mne-tools/mne-python/jobs/122007273

try running :

make test-fast

closed 2016-04-12 by jjnurminen