mne-tools/mne-python #3089: extract_label_time_course mode='pca_flip' has NaNs depending on label thresh

kdoelling1919, 2016-03-31:
I am trying to extract the time course from label created using mne.stc_to_label. The behavior seems to change depending on the threshold I used in the stc. The relevant piece of code I have written below. The code allows me to view a plot of the source estimate and choose a threshold as a percentage from maximum (i.e val = 70 -> threshold is .7*np.max(stc.data)). The trouble is that depending on my choice of val I can get NaNs: 70 works, 60 works but 62 creates NaNs in a few of the samples. 90 has NaNs and 75 gets an 'SVD did not converge' error. Also, both 53 and 6 work fine, in case you think its about the singles digit....

I just pulled the most recent mne version from github. My source estimate has no NaNs in it. I guess there is a problem with the labels? or in the extract_label_time_course function itself. Has anyone come across anything like this issue before?

Thanks in advance!
Keith

```
    #------- Set things up
    # read in stc and src files
    stc = mne.read_source_estimate(stcfile,subject=subject)
    inv = read_inverse_operator(invfile)
    src = inv['src']
    
    # master copy before editing
    master_stc=stc.copy()
    
    # pick peak around some time window and get average response
    pk=stc.get_peak(tmin=tlim[0],tmax=tlim[1])
    tmin=pk[1]-winsize/2.
    tmax=pk[1]+winsize/2.
    win_stc = stc.crop(tmin,tmax).mean()

    # find best clims and plot
    clim = np.max(win_stc.data)
    clims = (0*clim,.5*clim,.9*clim)
    brain = win_stc.plot(hemi='split',surface='white',subject=subject,subjects_dir=freesurfer_dir,
                        clim=dict(kind='value',lims=clims),transparent=False,alpha=.7)
    
    #------ create labels and inspect
    # enter a percentage for a threshold = val% * max response
    val=raw_input('Enter a percentage from 0-100: ')
    if not val:
        val = '70'
        print('Fine! Be that way... I''ll try 70...')
    while val:
        brain.remove_labels(hemi='lh')
        brain.remove_labels(hemi='rh')
        # check user input
        try:
            val=int(val)*.01
        except Exception:
            val = raw_input('Must enter a valid integer from 0-100!!!: ')
            continue
        # create label given this threshold
        print('Trying %f' % val)
        win_stc_label = win_stc.copy()
        # turn everything below threshold to 0
        win_stc_label.data[win_stc.data<val*np.max(win_stc.data)]=0
        # create the labels for both left and right hemispheres
        func_labels=mne.stc_to_label(win_stc_label,src,smooth=True,connected=False,
                                                        subjects_dir=freesurfer_dir)

        # add label to plot
        for i in xrange(len(func_labels)):
            brain.add_label(func_labels[i],color='b',borders=False,hemi=func_labels[i].hemi,alpha=.4)
        # if satisfied, move on. If not, try another percentage
        val=raw_input('If all good, press enter, else, more numbers please!')

    #--- extract time course from label using pca_flip and check for NaN
    pca_func=master_stc.extract_label_time_course(func_labels,src,mode='pca_flip')
    print(np.isnan(pca_func).any())
```

agramfort, 2016-04-01:
can you replicate the pb with sample dataset? it would make our life easier
to debug.

kdoelling1919, 2016-04-01:
I tried running the code on the sample dataset but I can't reproduce the problem. Seems to work fine in this case. Is there something I should check for in my source estimates or source space that could be causing this? If you like, I can send the *.stc and inv.fif file that do create the problem.

Eric89GXL, 2016-04-01:
If you can share them it might help

kdoelling1919, 2016-04-01:
Files are a bit too large to be uploaded to github repository. Here's a dropbox link:
https://www.dropbox.com/sh/0ec6q2lldjljbh2/AAALIwlLGlyw9GaxU0WY-BbYa?dl=0

agramfort, 2016-04-02:
please share a complete gist that replicates the pb. Just copy pasting your
code does not work.

Eric89GXL, 2016-07-18:
Closing due to non-response, but @kdoelling1919 please reopen if you can provide a gist to replicate

closed 2016-07-18 by Eric89GXL