mne-tools/mne-python #3094: scipy.linalg error with maxwell_filter

szorowi1, 2016-04-02:
Hi all,

Following up on an issue we've been running into when using the maxwell_filter function from mne 0.11. Attached are the files and code necessary to recreate the issue. 

[github_code.txt](https://github.com/mne-tools/mne-python/files/200990/github_code.txt)
[hc021_arc1_motion_hpi.txt](https://github.com/mne-tools/mne-python/files/200991/hc021_arc1_motion_hpi.txt)
[hc021_arc2_motion_hpi.txt](https://github.com/mne-tools/mne-python/files/200992/hc021_arc2_motion_hpi.txt)
[hc021_arc3_motion_hpi.txt](https://github.com/mne-tools/mne-python/files/200993/hc021_arc3_motion_hpi.txt)

Because the raw file is quite large, I've stored in a dropbox link here:

raw pt. 1: https://www.dropbox.com/s/5xnr8jn0e8dvwos/hc021_arc_rer_raw.fif?dl=0
raw pt. 2: https://www.dropbox.com/s/j9qset1pzzn95w8/hc021_arc_rer_raw-1.fif?dl=0

Let me know if there is anything else I can provide!

The error itself is posted below.

> Traceback (most recent call last):
>   File "motion_correction_flat.py", line 65, in <module>
>     raw = maxwell_filter(raw, st_duration=st_duration, destination=trans.mean(axis=0), verbose=True)
>   File "<string>", line 2, in maxwell_filter
>   File "/homes/5/szoro/.local/lib/python2.7/site-packages/mne/utils.py", line 548, in verbose
>     return function(*args, **kwargs)
>   File "/homes/5/szoro/.local/lib/python2.7/site-packages/mne/preprocessing/maxwell.py", line 404, in maxwell_filter
>     t_proj = _overlap_projector(orig_in_data, resid, st_correlation)
>   File "/homes/5/szoro/.local/lib/python2.7/site-packages/mne/preprocessing/maxwell.py", line 1237, in _overlap_projector
>     overwrite_a=True, mode='economic', **check_disable)[0].T
>   File "/homes/5/szoro/.local/lib/python2.7/site-packages/scipy/linalg/decomp_qr.py", line 142, in qr
>     overwrite_a=overwrite_a)
>   File "/homes/5/szoro/.local/lib/python2.7/site-packages/scipy/linalg/decomp_qr.py", line 20, in safecall
>     ret = f(*args, **kwargs)
> ValueError: failed to create intent(cache|hide)|optional array-- must have defined dimensions but got (0,)

szorowi1, 2016-04-02:
In response to Eric's request:

cat /proc/cpuinfo | grep avx | wc -l
24
cat /proc/cpuinfo | grep sse4_2 | wc -l
24
cat /proc/cpuinfo | grep sse4_1 | wc -l
24

Eric89GXL, 2016-04-02:
Can you give mne.sys_info() output too?

szorowi1, 2016-04-03:
Apologies, but is that an attribute of mne? I'm getting an error running mne.sys_info()

Eric89GXL, 2016-04-03:
Are you on latest master? It should be there.

If not, it would be good to know what Python you're running (anaconda,
system, etc.) and numpy and SciPy versions

szorowi1, 2016-04-05:
I thought I was, but evidently not. I'm running Anaconda. Numpy version 1.10.4. Scipy version 0.16.0

Eric89GXL, 2016-04-05:
That function was added only a month or two ago, if it matters.

Eric89GXL, 2016-04-05:
> from mne 0.11

Although from your previous post it sounds like you are on 0.11, not master?

szorowi1, 2016-04-05:
Just checked: I am on 0.11.

Eric89GXL, 2016-04-05:
The weird thing is that your `scipy` is imported from `.local/lib/python2.7/site-packages/scipy/linalg/decomp_qr.py`. I don't think that's usually where Anaconda would install it. I wonder if perhaps some program ended up doing `pip install scipy` and installing it to your system package location instead. What does `which python` say?

szorowi1, 2016-04-05:
/usr/pubsw/packages/python/anaconda/bin/python

Eric89GXL, 2016-04-05:
I suspect if you do `ls -d /usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/sci*` you'll see some `scipy` packages there, probably with version 0.16?

What does `python -c "import scipy; print(scipy); print(scipy.__version__)"` say?

szorowi1, 2016-04-05:
<module 'scipy' from '/homes/5/szoro/.local/lib/python2.7/site-packages/scipy/__init__.pyc'>
0.16.0

Eric89GXL, 2016-04-05:
I'm a bit stumped. You could try:
```
conda update conda
conda install mkl
```
And see if that updates anything. You could also do:
```
conda update scipy
```
But that could possibly break something.

szorowi1, 2016-04-05:
I'm not sure if this helps explain, but I'm at the Martinos center. So the anaconda distribution is sourced through the center's install; my MNE is sourced through a local install given that their time install is still on 0.9

Eric89GXL, 2016-05-11:
@szorowi1 any luck fixing this one, or are you still stuck?

Eric89GXL, 2016-05-13:
Closing since I suspect this is a packaging issue that has been resolved (?), but feel free to reopen if I'm mistaken

closed 2016-05-13 by Eric89GXL