mne-tools/mne-python #3097: Getting averaging categories from fiff file

jjnurminen, 2016-04-04:
Is there any facility for getting averaging categories from a fiff file? One can parse raw.info['acq_pars'], but it's not that user-friendly.

This info would be very useful e.g. for reaveraging data, so you could assign category labels automatically etc.

agramfort, 2016-04-04:
what is acq_pars ? I've never used it and it's typically empty for sample

Eric89GXL, 2016-04-04:
I haven't ever seen an `acq_pars` with useful data. If you have one to share, we could construct something like a `mne.events.parse_acq_pars(info)` function that returned a dict of `event_id` for use with `Epochs`.

Eric89GXL, 2016-04-04:
What causes that field to be populated? Use of Stim2, or setting which categories to average during Neuromag acquisition?

jjnurminen, 2016-04-04:
At least with Elekta/Neuromag systems, acq_pars contains the parameters that are set during acquisition setup. @Eric89GXL I'm kinda surprised that you haven't see any useful data there - don't you setup e.g. averaging categories in your measurement setup?

MaxFilter has an averager module which can just load a raw data file and reaverage all stimulus categories with one click. It takes the categories (trigger events, durations, labels etc.) from the fiff file itself, which is handy.

agramfort, 2016-04-04:
can you give an example of acq_pars value?

jjnurminen, 2016-04-04:
Full integration of the Elekta averaging logic with mne-python might require some thought, because Elekta distinguishes between events and averaging categories. Events correspond to trigger transitions, but categories might be associated with multiple events. (For example, a particular category might refer both to stimulus trigger event and a subject response event; an epoch will be added to the category only if both events occur within given time windows).

I could write a simple parser that at least would give the category labels, event codes and time windows.

jjnurminen, 2016-04-04:
@agramfort Here's an example of info['acq_pars'] from a TRIUX system. I think the fiff is actually from Eric's lab in Seattle :) As you see, there is a huge amount of variables there, basically all the acquisition related info.

[acqpars.txt](https://github.com/mne-tools/mne-python/files/202192/acqpars.txt)

agramfort, 2016-04-04:
thanks... I would need help to make sense of this :)

jjnurminen, 2016-04-04:
I'll write a simple parser for my own project. This will parse the most important info for the averaging categories defined in dacq. I would to make it into a pull request once I have time.

Eric89GXL, 2016-04-04:
I don't set up online averaging, but other people at UW must do it :)

jjnurminen, 2016-04-06:
mne.find_events seems to support only trigger transitions 0->x, right? Elekta systems (at least TRIUX) supports arbitrary bit transitions as events. So find_events would also need to be adapted.

dgwakeman, 2016-04-06:
@jjnurminen, no you can do any transition with it: check the ```consecutive``` input.

jjnurminen, 2016-04-06:
@agramfort oh, I see. But I think it's still more limited than TRIUX allows. For example, in TRIUX you can define that 1->3 is an event, but 2->3 is not, or vice versa. (find_events would report both as event id 3).

jjnurminen, 2016-04-06:
To fully implement TRIUX logic would require:

-considering both the pre-stimulus and post-stimulus trigger state (e.g. 1->3 and 2->3 should be recorded as different events)
-implementing bitwise masking separately for pre-stimulus and post-stimulus state (e.g. ignore bit 1 pre-stimulus but consider it post-stimulus)

Eric89GXL, 2016-04-06:
> find_events would report both as event id 3

The third column would be 3 in both cases, but the second column would not. So you could write a line of `numpy` to switch it to some other event_id like:
```
events[np.logical_and(events[:, 1] == 2, events[:, 2] == 3), 2] = 23
```

> implementing bitwise masking separately for pre-stimulus and post-stimulus state

I think you could accomplish bit-masking using a line or two of `numpy`. The idea is that we don't offer all of the functionality ourselves with `mne` functions if they can be accomplished fairly clearly using a few `numpy` steps. Make sense?

Eric89GXL, 2016-04-06:
If you agree about how to deal with these, we should *really* update the docs to make it clear how to accomplish these things. Maybe an entry in the FAQ would be a good start. @jjnurminen if you have time it would be an excellent contribution as you work them out, especially since you're probably most familiar with how Triux handles events.

dgwakeman, 2016-04-06:
Yeah, it could be something like:
How do I average like a Neuromag system? All Vectorviews average the same way.

jjnurminen, 2016-04-06:
@Eric89GXL Makes sense, except I'm not sure how the pre/post-event bitmasking could be handled outside find_events. It cannot be done before feeding the trigger channel to find_events, since you first need to know where the events are.

Eric89GXL, 2016-04-06:
I can't picture it. Can you give a small example of what STI101 values you'd have, and what output you'd want that you can't currently get?

jjnurminen, 2016-04-06:
@Eric89GXL oops, sorry. One would just get all transitions first with find_events, get both the pre/post event trigger values and then do the masking afterwards. Should work.

Eric89GXL, 2016-04-06:
Great. When you have time it would still be good to have, say, 100 (fake) samples of a trigger channel, plus what you could get with Triux's event options, then how you'd get the corresponding events to be used with `numpy` and `mne`. It would become part of the docs eventually.

jjnurminen, 2016-04-08:
OK, I now have written:

-Elekta_averager class that gets the data acquisition (DACQ) parameters (events, categories, rejection limits etc. from Raw.info.

-few lines of numpy that transforms mne.find_events output so that the third column refers to DACQ events rather than trigger codes, taking the bitmasks into account. As noted before, a single trigger code may correspond to multiple DACQ events, which can be binary coded into the third column.

-some further numpy that transforms the DACQ event list into a list of 'averaging triggers'. This needs consideration of both the reference event and required event. The final trigger list refers to averaging categories, i.e. the line [14990 0 1] would mean that category #1 should be averaged at time 14990 (=all conditions of the category are fulfilled at that time).

I think this actually fits rather nicely into the logic of mne-python. The final trigger list can be directly fed to mne.Epochs, and the corresponding event_id can be easily generated. Thus one gets mne.Epochs corresponding to one TRIUX averaging category.

jjnurminen, 2016-04-08:
Of course this needs some testing first, especially with more complicated averaging setups :)
But eventually I hope the Elekta_averager could be mergeable, and the code for transforming the mne-python event list could be included in that class.

Eric89GXL, 2016-04-08:
Usually we prefer functions to classes when possible. Could it be a function that took a set of input events (e.g., from `read_events`) and a few other parameters to accomplish this?

If not, what does the current API look like?

jjnurminen, 2016-04-08:
Classes seemed the most natural to contain the averager parameters and helper methods.
The averager has a lot of general parameters, and it contains a lot of events and averaging categories, which in turn have their own parameters.
Right now I can do:
```
eav = Elekta_averager(raw.info['acq_pars'])
<Elekta_averager | Categories: 32 Events: 32>

eav.magmax    # 1e-13
eav.magslope  # etc.

ev = eav.events[1]
<Elekta_event | name: Event1 comment: 'Event # 1' newbits: 1 oldbits: 0 newmask: 63 oldmask: 63 delay: 0>

cat = eav.categories['Visual']
<Elekta_category | comment: Visual event: 1 reqevent: 0 reqwhen: 1.0 reqwithin: 0.0 start: -0.1 end: 0.9>
```

That's just for organizing the averager info. The methods for converting find_events output would fit naturally as classmethods but they could also be somewhere else, of course.

jjnurminen, 2016-04-09:
Maybe Elekta_event and  Elekta_category could be just dicts inside the Averager class, if proliferation of classes is a concern. But classes are easier to expand when some new functionality is needed.

agramfort, 2016-04-09:
jussi please share your code in a PR with a demo example code

we'll discuss the API on the diff via github

jjnurminen, 2016-04-09:
Yep, it'll be coming sometime later. I'll first do some of the doc tasks.

Eric89GXL, 2016-04-13:
I finally looked through the `acq_pars` for a file that had them. Yikes. I see why a class would be useful.

I'm +1 on adding a class for this, probably in `mne.events`. @agramfort WDYT?

@jjnurminen how about we add a phantom dataset (preferbly with very few epochs) to the `testing` dataset? I think that's where I saw the `acq_pars`.

jjnurminen, 2016-04-14:
Hi @Eric89GXL 

Phantom dataset is a good idea IMO. It should preferably test all of the features (bitmasking, conditional events, delays etc.) and thus would need a pretty complicated averaging setup. (BTW, I thought Elekta/Neuromag systems (NM122, Vectorview, TRIUX) always write dacq_pars, so every fiff file should have it).

Yesterday I sent a weighted averaging script to Samu that uses the averager class. Maybe you can check out how it works. I'll make it into a PR later.

agramfort, 2016-04-14:
>
> I'm +1 on adding a class for this, probably in mne.events. @agramfort
> <https://github.com/agramfort> WDYT?
>
why not. If you can propose an API i'm curious

Eric89GXL, 2016-04-14:
@agramfort [look above] (https://github.com/mne-tools/mne-python/issues/3097#issuecomment-207610883)

Eric89GXL, 2016-04-14:
> BTW, I thought Elekta/Neuromag systems (NM122, Vectorview, TRIUX) always write dacq_pars, so every fiff file should have it

I just checked and it looks like the ones I've been recording do indeed have it. We didn't use any online averaging during acquisition, though, so it's probably useless.

Although the `sample` dataset *does* have it, the `sample_audvis_trunc_raw.fif` in the `testing` dataset does not. This is probably because we wrote that truncated file a long time ago and maybe didn't save `acq_pars` properly then. We could update it to have the information, then we don't even need to add a file (unless we want really complete support for all event variants). @jjnurminen can you see how useful `sample_audvis_raw.fif` in the `sample` dataset would be?

jjnurminen, 2016-04-14:
> We didn't use any online averaging during acquisition, though, so it's probably useless.

Not 100% sure what you mean here, but the averaging setup is very useful regardless of whether you do online averaging. Nowadays offline averaging is more common, but usually the DACQ averaging parameters are used for that too. (Indeed it's much more difficult to do offline averaging with Elekta software, if you haven't set up the DACQ categories before the measurement).

> can you see how useful sample_audvis_raw.fif in the sample dataset would be?

Sure, I can check it out. I doubt that it would use all the event/category features DACQ offers, but it can be a starting point.

Eric89GXL, 2016-04-14:
> Not 100% sure what you mean here, but the averaging setup is very useful regardless of whether you do online averaging

I mean that for many of the paradigms I've recorded, I've never told the acquisition software anything about what my triggers mean (and we actually use serial stamping sometimes), so I'm pretty sure `acq_pars` in most of my files don't have any useful information.

> I doubt that it would use all the event/category features DACQ offers, but it can be a starting point.

Let's see how close we get. My guess is that we won't get all that close, and we'll need to add a small phantom (or some other custom) dataset, but it's fine to double-test. In the meantime, hopefully working with the `sample` data you can come up with an example file in `examples/` somewhere to show it off using the `sample` dataset, in addition to developing some basic unit tests. I'd start with the example so that people can see what the API is like and play with it easily.

jjnurminen, 2016-04-14:
> I mean that for many of the paradigms I've recorded, I've never told the acquisition software anything about what my triggers mean (and we actually use serial stamping sometimes), so I'm pretty sure acq_pars in most of my files don't have any useful information.

I see. Well that suffices for simple paradigms. By serial stamping I guess you mean the videomeg timing triggers? We always have those running on in our lab, so they have to be masked out.

> I'd start with the example so that people can see what the API is like and play with it easily.

OK, makes sense.

Eric89GXL, 2016-04-14:
>  By serial stamping I guess you mean the videomeg timing triggers?

We use a custom stamping scheme, where we serially stamp a sequence of 4's and 8's (as a shifted binary code) to signify the trial type. This has to do with bugs in the stimulus delivery hardware that were encountered back in the day at MGH. Then we convert those triggers to meaningful decimal ones at the stimulus onset trigger (the 1 trigger for each trial) in custom software offline.

jjnurminen, 2016-04-15:
OK, that's pretty exotic :)
I looked at sample_audvis_raw.fif and it does have several different categories. I think it comes from a Vectorview system though, and the DACQ variables are somewhat different from TRIUX. I'll need to adapt the code for that.
I'll be travelling for the next two weeks but I'll try to submit the averager PR afterwards.

Eric89GXL, 2016-04-15:
No rush from our end. But yes we should probably support both VectorView and Triux if possible.

jjnurminen, 2016-05-03:
I put the averager code in events.py as Eric suggested. However I'm using the Epochs class and thus need to import it. I tried `from . import Epochs` and some variants at top of events.py, but the result is always ImportError due to circular imports. What would be the best way to resolve this?

agramfort, 2016-05-03:
can you make this code visible in a pull request?

jjnurminen, 2016-09-20:
Closing due to #3205 nearing completion

closed 2016-09-20 by jjnurminen