mne-tools/mne-python #3099: Epochs.next() docs don't match behavior

drammock, 2016-04-05:
[Docstring says here](https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L1423-L1424) that `Epochs.next()` returns an instance of `mne.epochs.Epochs`, but in fact it returns a `numpy.ndarray` (or a tuple of `ndarray, event_id` if `return_event_id=True`).  Personally I think the docs reflect how it _should_ work, but it's obviously easier to just change the docs to reflect what it _actually_ does.

agramfort, 2016-04-06:
indeed ... can you open a PR to fix the docstring?

drammock, 2016-04-06:
Will do (tomorrow, it's rather late in utc-8).

drammock, 2016-04-06:
@agramfort should I open a separate issue about the naming of `next()` vs. `iter_data()`, or is that a change that would be nice but will never happen?

Eric89GXL, 2016-04-06:
We want the `next` method to exist because we need it for iteration. I'm fine with making a new `iter_data` method basically just a clearer alias for `next`, if that's what you have in mind.

drammock, 2016-04-06:
well, what I have in mind is making `next` actually return an instance of `Epochs` (so that its methods are in place) and having a separate method like `iter_data` to iterate just the numpy arrays.  But that doesn't seem to be a popular view...  nobody is coming out and saying "we don't want to do that" but there seems to be ample support for the "just change the docs" approach.

agramfort, 2016-04-06:
I would say fix the doc for now. No rush for the rest. We have more urgent
matter to do. It's no blocker

PR #3104 (2016-04-06): DOC: Epochs.next and Epochs.iter_evoked
closes #3099 . ready for review @agramfort
closed 2016-04-06 by agramfort