mne-tools/mne-python #3100: MemoryError when plotting evokeds

drammock, 2016-04-05:
I've opened a matplotlib issue [here](https://github.com/matplotlib/matplotlib/issues/6269) but I am unsure whether it is actually their bug or a bug in mne.  An MWE including traceback is in [this gist](https://gist.github.com/drammock/babf1ca8f91fed444cab682d2c407eb6) and the data yielding the error is linked from the first gist comment.

agramfort, 2016-04-06:
I doubt you'll get an answer if it's actually a memory issue. Either you
track down extra allocations in matplotlib and suggest an improvement but
they will (i think) not consider it as a bug.

let us know what happens there

drammock, 2016-04-06:
I forgot to mention that my system's RAM monitor never moves above 25%
usage throughout these operations. I don't think it's a legitimate memory
error.

agramfort, 2016-04-06:
then it's certainly a bug :)

drammock, 2016-04-07:
FYI, after some further digging, the error does not occur if I use `mne.viz.evoked._plot_evoked(..., selectable=False)` so I'm guessing that the problem is somewhere in one of `_butterfly_onselect`, `_butterfly_onpick`, or `_butterfly_on_button_press`.   I also noticed that there is no way to disable `selectable` through public functions or methods.  That might be a bad idea, as it appears the interactivity (esp. the topoplot) will fail whenever digitization data is missing (which is likely for EEG-only data).

Note however that even though the plot now completes without errors, I wouldn't say it succeeds. Code and screenshot:

```
from mne.viz.evoked import _plot_evoked
import numpy as np
from matplotlib import pyplot as plt

evoked_list = np.load('evoked_list.npy')  # list of 12 mne.Evoked objects
# default args
args = dict(picks=None, exclude='bads', unit=True, show=True, ylim=None,
            xlim='tight', proj=False, hline=None, units=None,
            scalings=None, titles=None, axes=None, plot_type='butterfly',
            cmap=None, gfp=False, window_title=None, spatial_colors=False,
            set_tight_layout=True, selectable=True)
# override defaults
args.update(show=False, selectable=False)
# plot
nrow, ncol = (3, 4)
fig, axs = plt.subplots(nrow, ncol)
for ix, evoked in enumerate(evoked_list):
    ax = axs[ix // ncol, ix % ncol]
    args.update(evoked=evoked, axes=ax)
    _ = _plot_evoked(**args)
fig.show()
```
[image]

Eric89GXL, 2016-04-07:
Looks like you've hit another `matplotlib` bug, this time with `tight_layout`. If you comment out in the `viz` functions where `tight_layout` gets called, the plots will probably look better.

drammock, 2016-04-07:
@Eric89GXL true, if I update the standard args with `set_tight_layout=False` I do get slightly more reasonable-looking results:

[image]

drammock, 2016-04-07:
...in fact, doing that and then calling `plt.tight_layout()` manually is best of all:

[image]

jaeilepp, 2016-04-08:
It seems that disabling blit gets rid of the memory error, but interactive features don't really work well without it. Seems like ``evoked.plot`` needs a revision.

tacaswell, 2016-04-08:
My hunch at what is going wrong with blitting is that you are trying to copy out of a canvas at the wrong time and it is reporting a funny size (maybe `-1`?) which is getting cast to a uint _someplace_  which is rolling over to give you a memory error.

The place in mpl where we use blitting the most is in the animation code.  The way we do that is to grab the background on the `draw` event callback (which is the last thing to fire at the end of `Figure.draw`.

jaeilepp, 2016-04-26:
Now when I try running the gist, I get a ``KeyError: 'nchan'``. I think it has something to do with reverting the info PR. This should be fixed before 0.12. Taking a look now.

jaeilepp, 2016-04-26:
Okay, I think it's because the files were saved using the reverted info thing and using pickling. This is not a blocker for 0.12.

jaeilepp, 2016-04-29:
> FYI, after some further digging, the error does not occur if I use mne.viz.evoked._plot_evoked(..., selectable=False)

I can confirm this. Would it be okay to expose ``selectable`` param to the public functions?

Eric89GXL, 2016-04-29:
Fine by me

closed 2016-05-26 by jaeilepp

nbara, 2016-12-13:
FWIW I still get the Out of Memory error when using `selectable=False` with `spatial_colors=True`

jaeilepp, 2016-12-13:
I'm not able to reproduce the mem error with ``selectable=False``, however there is definitely some room for improvement in the visuals.