mne-tools/mne-python #3103: pick_types allows non-bool args for e.g. EEG

jona-sassenhagen, 2016-04-06:
I saw somebody doing this today:

`raw.pick_types(meg=False, eeg='grads')`

Which works (picking EEG channels).
Maybe a bit confusing, but not necessarily a bug ... Should we fix or keep?

Eric89GXL, 2016-04-06:
People could also do `pick_types(eeg=0)` which would evaluate the same way as False, and `pick_types(eeg=1)` which would eval as `True`, which I'm okay with. I guess we could strictly type check to be safe, though.

agramfort, 2016-04-06:
well I guess it's hard to catch all possible bugs but for such a common
function might we can make an effort to type check

jona-sassenhagen, 2016-04-06:
I can make a PR with something like
```
params = eeg, stim, ...
for param in params:
    if not isinstance(param, bool):
        raise ValueError(param + " must be of type ``bool``!")
```

agramfort, 2016-04-06:
yes but say also what is the type provided.

jaeilepp, 2016-04-07:
We have ``ref_meg='auto'`` as a default. I'm still unclear about what it means.

PR #3106 (2016-04-07): Check type==bool for pick_types
closes #3103

Surely my most interesting PR ever.
closed 2016-04-07 by agramfort