mne-tools/mne-python #3105: retyping channel when renaming it -  ecg not in channel list

tntdynamight, 2016-04-07:
Hi, 

I suggest adding a mne.channel.retype (or an automatic retyping to channel.rename_channels), especially for 'ecg' and 'eog'. I get those from biochannels, and can do `raw.pick_channels`; but without the correct type I can't use `mne.pick_types`or `ica_find_bads_ecg`. 

I retyped them manually by digging into the code belly and adding: 
```
raw.info['chs'][raw.info['ch_names'].index('ecg')]['kind']=402
raw.info['chs'][raw.info['ch_names'].index('eog')]['kind']=202
```

but I guess it would solve people a lot of time not having to find and go through the numeric channel identities. Also they may warrant a listing in the manual / API (section mne.channel), or maybe I overlooked that?

agramfort, 2016-04-07:
do you know the set_channel_types methods?

eg for raw:

http://martinos.org/mne/dev/generated/mne.io.Raw.html#mne.io.Raw.set_channel_types

tntdynamight, 2016-04-07:
Ah, indeed I overlooked it, though I did scan that page; my apologies. 

Should I turn to the mailing list first in general? 
Thanks, 
tnt

agramfort, 2016-04-07:
yes mailing list for such questions is better.

close the issue if we're good

closed 2016-04-07 by tntdynamight