mne-tools/mne-python #3112: Bem.fif is too big to be saved - how to implement split file

ualsbombe, 2016-04-11:
Dear Developers

I recently tried to create a BEM using the mne-python code, but I ran into the issue that it is too big to save, and there is no automatic splitting for it.

I guess I could use a source space with fewer sources in it this particular time, 

Below is the output from mne.write_bem_solution()

Best

Lau

> 3 BEM surfaces found
>     Reading a surface...
> [done]
>     Reading a surface...
> [done]
>     Reading a surface...
> [done]
>     3 BEM surfaces read
> Approximation method : Linear collocation
> 
> Three-layer model surfaces loaded.
> Computing the linear collocation solution...
>     Matrix coefficients...
>         head (10242) -> head (10242) ...
>         head (10242) -> outer_skull (10242) ...
>         head (10242) -> inner_skull (10242) ...
>         outer_skull (10242) -> head (10242) ...
>         outer_skull (10242) -> outer_skull (10242) ...
>         outer_skull (10242) -> inner_skull (10242) ...
>         inner_skull (10242) -> head (10242) ...
>         inner_skull (10242) -> outer_skull (10242) ...
>         inner_skull (10242) -> inner_skull (10242) ...
>     Inverting the coefficient matrix...
> IP approach required...
>     Matrix coefficients (homog)...
>         inner_skull (10242) -> inner_skull (10242) ...
>     Inverting the coefficient matrix (homog)...
>     Modify the original solution to incorporate IP approach...
>         Combining...
>         Scaling...
> Solution ready.
> BEM geometry computations complete.
> Traceback (most recent call last):
> 
>   File "<ipython-input-4-4e2811eb82a3>", line 4, in <module>
>     mne.write_bem_solution(save_path + bem_solution_name, bem_solution)
> 
>   File "/home/lau/src/mne/mne/bem.py", line 1416, in write_bem_solution
>     bem['solution'])
> 
>   File "/home/lau/src/mne/mne/io/write.py", line 135, in write_float_matrix
>     check_fiff_length(fid)
> 
>   File "/home/lau/src/mne/mne/io/write.py", line 270, in check_fiff_length
>     raise IOError('FIFF file exceeded 2GB limit, please split file or '
> 
> IOError: FIFF file exceeded 2GB limit, please split file or save to a different format
> _

agramfort, 2016-04-11:
don't use 10242 vertices per layer. Using half will be enough and will fix
the problem.

dengemann, 2016-04-11:
You really don't need a BEM with so many points, you can do this for the
source space instead if you need it. The conductivity model can be
estimated based on fewer points then the dipole grid used for inverse
solutions.

ualsbombe, 2016-04-11:
Ok, thanks for the input - I'll make the source space smaller.

Lau

Eric89GXL, 2016-04-11:
Reopened and re-titled. This is the second time this has come up, so we should probably add a warning in the `bem` code that 5120 should be enough, and also in the writing function if too many data points exist to save.

agramfort, 2016-04-11:
+100

closed 2016-07-04 by agramfort