mne-tools/mne-python #3113: nosetests attempts ftp download

jjnurminen, 2016-04-11:
Hi,
I tried running nosetests via make, but it fails attempting a ftp download. I'm behind a hospital firewall and 
cannot really use ftp (http downloads work fine via a proxy). Any suggestions?

If I only change documentation (docstrings and/or .rst), is it enough to run make test-doc?

agramfort, 2016-04-11:
yes it should be good. We'll polish it if needed

Eric89GXL, 2016-04-11:
Does it get to running the tests, and the file-downloading test fails? Or does it try to download a dataset *before* even running the tests? If it's the former, just ignore the failing download test -- the other tests are still valid. If it's the latter, you can run `nosetests -a '!slow_test' mne` (which is the second part in the Makefile).

You probably also want to run `nosetests mne/tests/test_docstring_parameters.py` to test some aspects of docstring formatting.

jjnurminen, 2016-04-11:
It was the former, but somehow the failed ftp download brought the whole thing down and the tests stopped running right there. I lost the console output, but I'll try it again.

jjnurminen, 2016-04-12:
[output.txt](https://github.com/mne-tools/mne-python/files/214502/output.txt)

Attached console output.

agramfort, 2016-04-12:
ERROR: Test file downloading is unrelated.

FAIL: Test Raw compensation by comparing with MNE is more problematic.
Maybe your sample dataset is not up to date?

Eric89GXL, 2016-04-12:
Probably the testing dataset needs an update.

Eric89GXL, 2016-04-14:
@jjnurminen okay to close this one? I think a test failing about FTP download is the right behavior if you can't actually download using FTP. If you really want to avoid seeing these failures, we could add a mne config like `MNE_SKIP_FTP_TESTS` that you and anyone else who can't use FTP could set to true.

jjnurminen, 2016-04-14:
Nice! I'll try it tomorrow. Closing now.

PR #3129 (2016-04-20): MRG: Allow skipping FTP tests
Closes #3113.

@jjnurminen you should be able to do `mne.set_config('MNE_SKIP_FTP_TESTS', 'true')` and have it not run those tests. Can you try it?
closed 2016-04-14 by jjnurminen