mne-tools/mne-python #3116: ENH: weighted averaging of epochs

jjnurminen, 2016-04-12:
There doesn't seem to be a facility for weighted averaging of epochs.
As an API, I propose adding a weights= parameter to Epochs.average(). It would take a list or 1-dim numpy array of weights, length equal to number of epochs.
I have already implemented weighting and can provide a PR at some point if the API sounds cool.

jona-sassenhagen, 2016-04-12:
Epoch separately and run mne.combine_evokeds with weights?

jjnurminen, 2016-04-12:
@jona-sassenhagen this would provide a way to weigh each epoch individually.

jona-sassenhagen, 2016-04-12:
I didn't expect that anyone would want to do that, but even then, couldn't you do the same by iterating over epochs to get each as an evoked and doing combine_evokdes?

jjnurminen, 2016-04-12:
I think it's not that uncommon to do weighting based on per-epoch SNR.
What I'm doing is weighting based on continuous HPI signal, i.e. less weight to epochs where the head is further away from the helmet.
The combine_evokeds method may work, but 1) a bit hackish IMO, didn't even occur to me 2) what if there are thousands of epochs? Weighting the data matrix before averaging is really quick and doesn't use much memory.

jona-sassenhagen, 2016-04-12:
If that's a common task, it totally makes sense to me.

agramfort, 2016-04-12:
this is really related to the robust averaging discussion...

my feeling is that it's a really advanced use case but why not. Can you
show some evidence of the benefit of this? A paper? Graphs?

jjnurminen, 2016-04-12:
What is robust averaging?

It's true that it might be somewhat advanced, but isn't mne for advanced stuff? ;)
I will provide some evidence later.

agramfort, 2016-04-12:
>
> What is robust averaging?
>

https://github.com/mne-tools/mne-python/issues/3044

> It's true that it might be somewhat advanced, but isn't mne for advanced
> stuff? ;)
> I will provide some evidence later.
>
great

Eric89GXL, 2016-04-12:
I don't think the overhead for iterating over a thousand epochs to
construct the weighed average will be so bad, have you tried it to see what
the performance penalty is like compared to the vectorized version? It
should only be a couple of lines to do per-epoch weighting.

jjnurminen, 2016-04-12:
So iterate over epochs, create data matrix corresponding to weighted average and then call mne.EvokedArray? That sounds reasonable.

agramfort, 2016-04-12:
yes that does the job

jjnurminen, 2016-04-12:
Ok, since the use case may be rather rare and Eric's suggestion will work, I'll close this now.

closed 2016-04-12 by jjnurminen