mne-tools/mne-python #3117: wrong layout for planar grad in plot_projs_topomap

agramfort, 2016-04-12:
Looking at https://mne-tools.github.io/dev/auto_examples/visualization/plot_ssp_projs_topomaps.html I see issues

- Some projs are duplicated as they are present multiple times in the object. That's not the biggest issue but still we should fix it.

- the name of sample_audvis_ecg_proj.fif does not match our naming conventions. 

See the message  "This filename (/home/ubuntu/mne_data/MNE-sample-data/MEG/sample/sample_audvis_ecg_proj.fif) does not conform to MNE naming conventions. All projection files should end with -proj.fif or -proj.fif.gz"

- finally and more importantly the plots for planar grads is not correct. There is no way it can be negative when plotting a topomap for a planar grad ssp vector. It should display the magnitude of the SSP.

any volunteer?

jaeilepp, 2016-04-12:
I can take a look.

agramfort, 2016-04-12:
cool thanks

closed 2016-04-15 by agramfort