mne-tools/mne-python #3118: Error in ica.plot_component

teonbrooks, 2016-04-12:
Is there a particular reason or has anyone had problems with `ica.plot_components()`. I'm getting an error with the `is_bbox`. I am guessing that the bounding box is not being constructed for my channel array. This also happens when I specify the layout for the system.

In [51]: ica.plot_components()
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-51-31baf63ee5ff> in <module>()
----> 1 ica.plot_components()

/Applications/packages/mne-python/mne/preprocessing/ica.py in plot_components(self, picks, ch_type, res, layout, vmin, vmax, cmap, sensors, colorbar, title, show, outlines, contours, image_interp, head_pos)
   1384                                    outlines=outlines, contours=contours,
   1385                                    image_interp=image_interp,
-> 1386                                    head_pos=head_pos)
   1387 
   1388     def plot_sources(self, inst, picks=None, exclude=None, start=None,

/Applications/packages/mne-python/mne/viz/topomap.py in plot_ica_components(ica, picks, ch_type, res, layout, vmin, vmax, cmap, sensors, colorbar, title, show, outlines, contours, image_interp, head_pos)
    748                                       show=show, outlines=outlines,
    749                                       contours=contours,
--> 750                                       image_interp=image_interp)
    751             figs.append(fig)
    752         return figs

/Applications/packages/mne-python/mne/viz/topomap.py in plot_ica_components(ica, picks, ch_type, res, layout, vmin, vmax, cmap, sensors, colorbar, title, show, outlines, contours, image_interp, head_pos)
    799     tight_layout(fig=fig)
    800     fig.subplots_adjust(top=0.95)
--> 801     fig.canvas.draw()
    802     plt_show(show)
    803     return fig

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_qt5agg.pyc in draw(self)
    146         # causes problems with code that uses the result of the
    147         # draw() to update plot elements.
--> 148         FigureCanvasAgg.draw(self)
    149         self._priv_update()
    150 

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_agg.pyc in draw(self)
    467 
    468         try:
--> 469             self.figure.draw(self.renderer)
    470         finally:
    471             RendererAgg.lock.release()

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/artist.pyc in draw_wrapper(artist, renderer, *args, **kwargs)
     57     def draw_wrapper(artist, renderer, *args, **kwargs):
     58         before(artist, renderer)
---> 59         draw(artist, renderer, *args, **kwargs)
     60         after(artist, renderer)
     61 

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/figure.pyc in draw(self, renderer)
   1015         if self.get_tight_layout() and self.axes:
   1016             try:
-> 1017                 self.tight_layout(renderer, **self._tight_parameters)
   1018             except ValueError:
   1019                 pass

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/figure.pyc in tight_layout(self, renderer, pad, h_pad, w_pad, rect)
   1661                                          renderer,
   1662                                          pad=pad, h_pad=h_pad, w_pad=w_pad,
-> 1663                                          rect=rect)
   1664 
   1665         self.subplots_adjust(**kwargs)

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/tight_layout.pyc in get_tight_layout_figure(fig, axes_list, subplotspec_list, renderer, pad, h_pad, w_pad, rect)
    350                                      subplot_list=subplot_list,
    351                                      ax_bbox_list=ax_bbox_list,
--> 352                                      pad=pad, h_pad=h_pad, w_pad=w_pad)
    353 
    354     if rect is not None:

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/tight_layout.pyc in auto_adjust_subplotpars(fig, renderer, nrows_ncols, num1num2_list, subplot_list, ax_bbox_list, pad, h_pad, w_pad, rect)
    129         tight_bbox_raw = union([ax.get_tightbbox(renderer) for ax in subplots])
    130         tight_bbox = TransformedBbox(tight_bbox_raw,
--> 131                                      fig.transFigure.inverted())
    132 
    133         row1, col1 = divmod(num1, cols)

/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/transforms.pyc in __init__(self, bbox, transform, **kwargs)
   1055         *transform*: a 2D :class:`Transform`
   1056         """
-> 1057         assert bbox.is_bbox
   1058         assert isinstance(transform, Transform)
   1059         assert transform.input_dims == 2

```
In [52]: ica.plot_components()
AttributeError: 'NoneType' object has no attribute 'is_bbox'

In [52]: Traceback (most recent call last):
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_qt5.py", line 427, in idle_draw
    self.draw()
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_qt5agg.py", line 148, in draw
    FigureCanvasAgg.draw(self)
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_agg.py", line 469, in draw
    self.figure.draw(self.renderer)
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/artist.py", line 59, in draw_wrapper
    draw(artist, renderer, *args, **kwargs)
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/figure.py", line 1017, in draw
    self.tight_layout(renderer, **self._tight_parameters)
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/figure.py", line 1663, in tight_layout
    rect=rect)
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/tight_layout.py", line 352, in get_tight_layout_figure
    pad=pad, h_pad=h_pad, w_pad=w_pad)
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/tight_layout.py", line 131, in auto_adjust_subplotpars
    fig.transFigure.inverted())
  File "/Users/teon/anaconda/lib/python2.7/site-packages/matplotlib/transforms.py", line 1057, in __init__
    assert bbox.is_bbox
AttributeError: 'NoneType' object has no attribute 'is_bbox'

If you suspect this is an IPython bug, please report it at:
    https://github.com/ipython/ipython/issues
or send an email to the mailing list at ipython-dev@scipy.org

You can print a more detailed traceback right now with "%tb", or use "%debug"
to interactively debug it.

Extra-detailed tracebacks for bug-reporting purposes can be enabled via:
    %config Application.verbose_crash=True
```

jona-sassenhagen, 2016-04-12:
I've also had bbox problems when plotting components with corrmap recently.

agramfort, 2016-04-13:
do you have the pb sample too? what backend are you using?

jaeilepp, 2016-04-13:
Any idea how to reproduce?

dengemann, 2016-04-13:
maybe matplotlib backend error? tried with different backends?

teonbrooks, 2016-04-13:
I'm using qt4 as the backend. I just got the same error with the osx backend too.
```
import mne
import os.path as op

data_path = mne.datasets.sample.data_path()
raw = op.join(data_path, 'MEG', 'sample', 'sample_audvis_raw.fif')

raw = mne.io.read_raw_fif(raw)
ica = mne.preprocessing.ICA(.9)
ica.fit(raw, decim=4)

ica.plot_components()
```

jona-sassenhagen, 2016-04-23:
The code Teon posted also errors out for me. It also blocks Corrmap. I think it might be related to ica.plot_components() calls with a specific number of components (below 20).

jona-sassenhagen, 2016-04-23:
I also get the error using `%matplotlib inline`.

jona-sassenhagen, 2016-04-24:
Some tests show that it's indeed about the number of components.

Whenever multiple lines are plotted, but one of the lines is not full, the error occurs. IE, when n_components > 5, but not divisible by 5. (1, 2, 3, 4, 5, 10, 20, 30, ... work, everything else doesn't work.)

Would be really cool if somebody could tackle this.

closed 2016-04-25 by Eric89GXL