mne-tools/mne-python #3120: fine tuning for plot sensors

dengemann, 2016-04-13:
After merging #3065 the next thing would be to add axis, picks and color parameters. Why is this? It facilitates creating educational plots of ROIS, e.g. like that:
[image]

teonbrooks, 2016-04-13:
that looks great

dengemann, 2016-04-13:
> that looks great

thanks, so far done with Python-(minus)MNE

jona-sassenhagen, 2016-04-19:
Yes I like it. This is used in many ERP publications.

jaeilepp, 2016-05-24:
This is now done in https://github.com/mne-tools/mne-python/pull/3237 although there is no color parameter. The color is automatically defined when you use an array as ``regions``. See if it satisfies you.

agramfort, 2016-05-24:
quick feedback after playing a bit with it

- regions seems like a misnomer as it can be a list of arrays that are not clusters / form regions

- regions='selection' would allow to see the automatic selections used in raw.plot

- raw.plot_sensors(ch_type='grad', regions='position')  # crashes on SPM dataset
# returns:
RuntimeError: Did not find any digitization points of kind FIFFV_POINT_EEG (3) in the info.

but this works

raw.plot_sensors(regions='position')

- raw.copy().pick_types(meg=False, eeg=True).plot_sensors(regions='position')

gives rather weird results.

Also see with grad or mag on sample. See the light blue dot at the right corner.

[image]

jaeilepp, 2016-05-25:
> regions seems like a misnomer as it can be a list of arrays that are not clusters / form regions

Any suggestion for a better name? All the other comments addressed.

jona-sassenhagen, 2016-08-28:
Can this be closed due to #3237 then?

agramfort, 2016-08-29:
I guess so.

closed 2016-08-29 by agramfort