mne-tools/mne-python #3121: ENH: default 2D proj of topomaps

kingjr, 2016-04-13:
Since the mood seems to be around sensor plotting and topo projections (https://github.com/mne-tools/mne-python/pull/3065, https://github.com/mne-tools/mne-python/issues/3078, https://github.com/mne-tools/mne-python/pull/3065), I think it would be great if we could add an option to project the sensor data onto a default head (projection, or in 3D) to get more plots: e.g.

[image]
[image]

I suspect this is already possible with a bit of hack.

dengemann, 2016-04-13:
+1000 would be excellent for template matching across datasets and subjects.

kingjr, 2016-05-01:
This is actually pretty hard to do it right for MEG, isn't it? I was thinking of using the 3D viz, do an inverse model, and apply the forward to virtual sensors made from the fsaverage mesh, but that seems a bit overkilling (and also, I don't manage to make these virtual sensors)... 

Any idea how we could procede?

teonbrooks, 2016-05-01:
can't you apply the `dev_head_t` to the MEG signal so that is on the head surface? is that what you're trying to do?

teonbrooks, 2016-05-01:
or `mne.forward.make_field_map` to project to head might be helpful

kingjr, 2016-05-01:
@teonbrooks indeed! Thanks, I'll check this out

kingjr, 2016-05-01:
Seems that all the functions are already in place, I was just no aware of them:

```
import os.path as op
import mne
from mne.forward import make_field_map
from mne.viz import plot_evoked_field

data_path = mne.datasets.sample.data_path()
subjects_dir = data_path + '/subjects'
trans_fname = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'
fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(fname, baseline=(None, 0), proj=True)[0]
evoked.pick_types(meg='grad')
maps = make_field_map(evoked, trans=trans_fname, subjects_dir=subjects_dir,
                      subject='sample', n_jobs=-1, meg_surf='head')
plot_evoked_field(evoked, maps, time=.1)
```

Shall I close?

kingjr, 2016-05-01:
related to this
1. Would it make sense to precompute the "maps" of the fsaverage subject, so as to gain time?
2. How can I plot the gradient of magnetic field (analogous to combined grads) on the head?

agramfort, 2016-05-02:
>
> 1. Would it make sense to precompute the "maps" of the fsaverage subject,
> so as to gain time?
>

it is really that slow? we could think of a joblib based caching system.

> 2. How can I plot the gradient of magnetic field (analogous to combined
> grads) on the head?
>

AFAIK currently you can't

teonbrooks, 2016-05-08:
@kingjr is the plan to have a topomap with option for head-oriented topography on a cartoon head?

kingjr, 2016-05-09:
@teonbrooks Well I thought the cartoon wouldn't be great. If we do it, might as well do it properly, no?

teonbrooks, 2016-05-09:
what would you use other than the default model?

kingjr, 2016-05-09:
> what would you use other than the default model?

I would aim for having default head, but this could be then changed to average head across subjects, or subject-specific heads.

Overall, I think it would make our papers more intuitive for naive readers if we plotted heads instead of flattened sensor arrays. But if we can't plot field gradients (tangential to the head surface), it won't be as useful as I had originally hoped because one would still have to mentally visualize where the dipole is likely to be coming from (i.e. in between the red and blue patches).

Eric89GXL, 2016-05-09:
Actually thinking about it more you can project to any virtual sensors you
like with the minimum norm code. So you could project to colocated
orthogonal pairs of arbitrarily placed planar gradiometers and then take
the norm of each pair if you want.

kingjr, 2016-05-09:
@Eric89GXL so we just have to find a way to make virtual sensors from the head mesh?

Eric89GXL, 2016-05-09:
Yeah you have to choose locations and normals

kingjr, 2016-05-10:
BTW, isn't the projection of sensor to scalp an ill-posed problem?

Eric89GXL, 2016-05-10:
We project to a minimum norm spherical source space and back out essentially

kingjr, 2016-05-10:
Well, I'm stuck. 

In `_make_surface_mapping` the virtual coils can still be gradiometers (at least their `type` matches those of neuromag grads), but I don't know where to look then. Do I need to modify `_do_self_dots`, `_do_surface_dots` and `_compute_mapping_matrix`. Any advice?

Eric89GXL, 2017-02-14:
> In _make_surface_mapping the virtual coils can still be gradiometers

@kingjr if you're interested in coming back to this... yes the *can* be gradiometers, but you can also put any other sensors you want there, e.g. point magnetometers.

closed 2023-09-25 by larsoner