mne-tools/mne-python #3124: interpolate_bads missing copy param; any others?

jona-sassenhagen, 2016-04-13:
I found one more method that should have, but doesn't have, a copy param.
Can anybody think of any others still missing theirs?

Otherwise, I'll do a PR ASAP.

dengemann, 2016-04-13:
sometimes I'm wondering why we introduced copy params. We had earlier introduced the .copy method. This is a potential API mess. Is there any case where inst.copy().method() would not do the job? As long as .method() returns self.

jona-sassenhagen, 2016-04-13:
1. consistency
2. I chain methods all the time, e.g. ica.fit(raw.filter(1, 20, copy=True)). This can be really comfortable IMO.
3. it's the most consistent way to allow the user to make it explicit if something should be done in-place or not. I know many people who're confused about when something is in-place and when not. `copy` for each method is the best way to get this straight.

dengemann, 2016-04-13:
I think the original idea was:

ica.fit(raw.copy().filter(1, 20))

This why every object has a .copy method. With this you should be able to express any copy/non-copy logic.

And from a core Python view methods typically operate in place on mutable data containers.

The only point would probably be delayed evaluation in loops, etc, when you would want to use copy as bool or key. 

```Python

for copy  in [True, False]:
    ica.fit(raw.filter(1, 20, copy=copy))

```

But why would you do that?

jona-sassenhagen, 2016-04-13:
Well we have always had a copy param for *some* methods. It's more consistent to have it across the board, and use that, instead of wildly mixing the two.
Also, checking what copy defaults to is a fast and clear way of checking if something operates in-place per default :)

teonbrooks, 2016-04-13:
I kinda agree that there should be one way but we currently have two. I am in favor of the copy method but this is not something that has been promoted. if we truly have it for all our objects, then that would be the best approach IMO. it still enables chaining and it's explicit. methods typically work on objects, so the introduction of the copy params (not this case but the first one) may have not been necessary

agramfort, 2016-04-14:
jona if ica.fit(raw.copy().filter(1, 20)) works can you live with it?

dengemann, 2016-04-14:
The more I think about it the more I feel that this would be worth a global deprecation of copy parameters while making sure that .copy() + chaining works everywhere.

dengemann, 2016-04-14:
the nice thing about copy is that it can be inserted as a step in any chaining pipeline and we never have to care about adding it to some "downstream" function. I think that was also the initial idea.

Eric89GXL, 2016-04-14:
> The more I think about it the more I feel that this would be worth a global deprecation of copy parameters while making sure that .copy() + chaining works everywhere.

Unfortunately in the last couple of months we've had a couple of PRs that did the opposite, so this will now be harder :(

I'm okay with removing `copy=` kwargs in favor of `.copy()`. But we should get it done before 0.12 if we want it.

jona-sassenhagen, 2016-04-14:
We just added like 10 copy params.

And there are a handful of methods - e.g. equalize_event_counts - that default to copy=True. This would be a major overhaul.

dengemann, 2016-04-14:
> We just added like 10 copy params.

did not have time to read / think about it or went unnoticed, sorry.

> Unfortunately in the last couple of months we've had a couple of PRs that did the opposite, so this will now be harder :(

I know but I think for future maintainablility and doing things the one way this would indeed be a reasonable pain to accept. We will have a more consistent and simpler API.

Eric89GXL, 2016-04-14:
> I know but I think for future maintainablility and doing things the one way this would indeed be a reasonable pain to accept. We will have a more consistent and simpler API.

So it sounds like @dengemann and @teonbrooks are +1 on removing `copy` param from functions. I'm sitting around +0, @jona-sassenhagen I take it is -1. @agramfort WDYT?

I can probably work on the PR if we decide to do it. It has been too many weeks since I had a PR that touched a majority of the repo files :)

dengemann, 2016-04-14:
Talked to @agramfort, he would prefer the .copy pattern, which is why we
initially introduced it. Maybe @agramfort wants to say something. Let's not
rush over it.

jona-sassenhagen, 2016-04-14:
Well okay. How to handle the cases that default to copy=True?

Eric89GXL, 2016-04-14:
> Let's not rush over it.

Actually there is a little bit of time pressure. We're currently scheduled to release at the end of the month, and we (this cycle) introduced about 10 such `copy=` kwargs. So if we want to remove them, better to do it now than a month from now, where our deprecation burden will increase significantly.

jona-sassenhagen, 2016-04-14:
@kingjr was also a part of the discussion.

kingjr, 2016-04-14:
I'm also -1 on removing the copy param.

When working on subsets of the data, and don't want to either loose the whole object, or double memory allocation:

e.g.
```
for ch_type in ch_types:
    epochs_ = epochs.pick_types(ch_type, copy=True)
    gat.fit(epochs_)
```

same issue with `crop` for instance.

Eric89GXL, 2016-04-14:
> When working on subsets of the data, and don't want to either loose the whole object, or double memory allocation

All `copy=True` does in most functions is do `object.copy()` at the top, so it will double memory allocation. Are there cases where it does something smarter? In the example you cite with `epochs.pick_types`, the first line copies the object:

https://github.com/mne-tools/mne-python/blob/master/mne/channels/channels.py#L447

So it is equivalent memory-wise to do `epochs = epochs.copy().pick_types(ch_type, copy=False)`.

kingjr, 2016-04-14:
@Eric89GXL Ok, if that's the case I'm +1 for dropping the argument if it simplifies the API; we have a tendency to make pretty heavy functions, so might as well try to minimize this issue.

Overall, we still need deprecation warnings for those copy params that were introduced before the previous release.

Eric89GXL, 2016-04-14:
Yeah someone will need to determine which are new and which are old... I'm aware of the annoying-ness of this task, which is why I volunteered myself instead of some other poor soul :)

kingjr, 2016-04-14:
@dengemann @kaichogami We also have to think about the `Pipeline` design. Are we sure we're not going to need copy params there?

kaichogami, 2016-04-14:
If `.copy()` does everything a `copy=True` param would, I don't see any reasons to include it in `Pipleline`.

kingjr, 2016-04-14:
@kaichogami because in sklearn pipeline we can only pipe `predict`, `fit` and `score` methods.

jona-sassenhagen, 2016-04-14:
In case we decide on keeping it, I suggested in another thread that we could have a copy decorator, wrapping the function in ´´´inst = inst.copy()´´´. That would make this very API light actually.

jona-sassenhagen, 2016-04-14:
Oh and btw, as for this thread, interpolate_bads should still return self, regardless of copy or not.

Eric89GXL, 2016-04-14:
The only case I can think of that you could do before that you couldn't after removing `copy` kwarg would be if you could pass `kwargs=dict(copy=True)` to some function before, and now you can't. But that can be solved with a trivial wrapper function that calls `copy` then whatever method.

> That would make this very API light actually.

From a code-added standpoint, it is a little bit more DRY but otherwise pretty close -- `@copier` is one line just like `inst = inst.copy() if copy else inst` is currently. But keep in mind the docstrings mult also be added everywhere for the `copy` arg, too. `.copy()` overlaps with `copy=` kwarg, so ideally one of them would go away. `.copy()` is actually a bit more general, too, in the sense that if a user wants an object to play with themselves (arbitrary changes) they can get it immediately.

agramfort, 2016-04-14:
if we can stick to the copy method without breaking too much code i am
definitely +1. we need to make sure we always return self...

dengemann, 2016-04-14:
yes that is the right deal, let's make sure to always return self, advocate
the use of .copy and we have something very light, future-friendly and
general.

Eric89GXL, 2016-04-18:
> Oh and btw, as for this thread, interpolate_bads should still return self, regardless of copy or not.

@jona-sassenhagen it looks like `interpolate_bads` already returns self, where does it not for you?

jona-sassenhagen, 2016-04-19:
>@jona-sassenhagen it looks like interpolate_bads already returns self, where does it not for you?
Hm ... no idea what I was thinking here.

PR #3139 (2016-04-28): MRG: Unify copy in methods
This PR aims to deprecate `copy=` parameters in methods (while leaving `copy` parameters alone in functions).

Closes #3124.

One outstanding problem is the few methods (e.g., `raw.crop`) that have `copy=True` by default, users will have slightly awkward interactions with for a cycle. The behavior is currently:

1. For current `copy=False` default functions:
    1. If `copy=None` (default) or `copy=False`, interpret as `copy=False`
    2. If `copy=True`, give `DeprecationWarning` and interpret as `copy=True`.
2. For current `copy=True` default functions:
    1. If `c [... cut]
closed 2016-04-19 by jona-sassenhagen