mne-tools/mne-python #3125: Feature: Obtain fold-wise scores in GAT

fraimondo, 2016-04-14:
I'm currently doing some analysis on several data and I came to the point I would like to get the SEM across fold scores from the GeneralizationAcrossTime object. 

The problem is that in `time_gen.py`, `_predict` we can find this:
```Python
    if fold > 0:
        if is_classifier(clf) and (predict_method == 'predict'):
            y_pred, _ = stats.mode(y_pred, axis=2)
        else:
            y_pred = np.mean(y_pred, axis=2, keepdims=True)
    y_pred = y_pred[:, :, 0]
```

That means that either mode or mean can be used based on the input. But there's no option of doing something different like SEM, or STD.

So far I've got to proposals:

1- Add an option to obtain the raw scores
2- Add an option to pass an aggregate function (or 'average' as default to use mean) that could also be None, meaning no aggregation, get the fold-wise scores and do whatever you want with that.

What do you think?

I can do it and make the corresponding test, but It seems a good idea to discuss it here.

@dengemann @kingjr

dengemann, 2016-04-14:
The second options sounds pretty intuitive to me.

kingjr, 2016-04-14:
I m also for the second option. Check that it doesn't break the aggregation
across cores.

dengemann, 2016-04-18:
Oh boy @kingjr I think we've had a really unnecessary bug in our code. The scoring just ignores the folds and goes across all trials. This is not equivalent to scoring fold wise and then averaging for most scores and essentially is not cross validation. @fraimondo on top of that reports difficulties in reproducing old findings, I think that must be the issue. This needs a fix.

kingjr, 2016-04-18:
Yes that s was i was saying , this is the departure from sklearn; it s not
a bug per se, the cross val must be applied at fit and predict stages, not
at scoring. Applying a CV at scoring only allows you to estimate the
folding error.

dengemann, 2016-04-18:
Well it is a bug as it breaks the contract with CV. We should reall fix
this and make a bug notice

kingjr, 2016-04-18:
OK. I would add a scoring option to do 'mean-across-folds' (what you
want?), 'per-fold' (what Fede needs, i.e. similar to sklearn), 'no-fold'
(current approach, which is n_fold faster, and still gives meaningful
results).

dengemann, 2016-04-18:
I think we should have as default what one would expect to get for cross
validation: score foldwise and average, the rest can be options and for
other measures than accuracy should throw a non-equivalence warning when
scoring all trials. If it approximately works or not is relatively
unimportant as we are using all conceptual support, it is simply not cross
validation and certainly not acceptable for us as a default.

kingjr, 2016-04-18:
Sure. Note that accuracy won't be equivalent either, since different folds
may have different number of trials.

The cross val here is applied across fit/predict, as it should. sklearn
users expect the cross val to also be applied for score. It's an API
decision which they use because their users often look at the SEM across
folds - which we don't. Again, I'm more than happy to get closer to sklearn
API, as long as we can keep the current (faster) scheme.

dengemann, 2016-04-18:
Yes that's another good point. So we should rather mark the current default
as an expert option with "proceed at your own risk if you know what you are
doing" tag if not abandon it. What do speed-gains mean if the result cannot
be interpreted? I would not call this an API decision, it's the common
sense about the semantics of cross-validation.

agramfort, 2016-04-18:
@kingjr what you did is what the iid=True option was doing in
cross_val_score
and it was removed as it is not what cross-validation scores are.

people should not be able to report numbers as cross-validation if it is
actually not a CV score. To me it is (close to a) bug.

kingjr, 2016-04-18:
I'm happy to call it a bug and change the defaut api but the results are
interpretable imo.

Which are the cases where one cant interpret scores across all cross
validated predictions in the context of second order stats that these
decoding functions are used ?

dengemann, 2016-04-18:
It's just that you cannot call it cross-validated AUC score any longer. It
is at best something unknown and does not comply with the properties that
have been shown for cross validation as an estimator of the out of sample
performance. Of course you get some output that maybe can be somehow used
in second order stats. But then you cannot say any longer that the second
order stats are second order stats of cross-validated discriminative
performance. This is not a situation in which you want to get. I hope that
the current behavior is not distorting things at least in a direction that
is more favorable to the researchers desired outcome. If we don't want
change everything here we should stick with at least a few basic contracts,
protocols and conventions. The fact that your computer does not crash and
you like the result is not a sufficient criterion for validity. And we did
not invent CV and the according practices, this is the result of research
in statistics. So for the scope of MNE I'd suggest to respect this.

agramfort, 2016-04-19:
@kingjr if you want to justify this non standard choice you need to
demonstrate
that it's actually better and valid. Do you have experimental evidence you
can
share for this?

kingjr, 2016-04-19:
Common Denis, the argument is not that it doesn't crash on my computer.

The reason d'etre of  CV is over fitting. Overfitting happens at the
fitting stage by definition, and doesn't bias the effect size estimate.

Scoring across all samples prevents estimating the fold error and thus the
estimation of the confidence interval of each score (there s no other way
around since the CV breaks the independence between samples, one can only
estimate the CI within a fold). This use case is incompatible with the
current implementation, and I m happy to correct for and warn on that.

I'll try to write the fix before the release.

dengemann, 2016-04-19:
> Common Denis, the argument is not that it doesn't crash on my computer.
Nope, scoring is part of the standard definition.

Well you know what I mean :)
As Alex and I highlighted a few times departing from standards (according
to which scoring is part of the CV which is not about overfitting but about
assessing your out of sample performance) is not what we want. We should
know better and have good reasons for it if we depart from such basic
notions. Call it bug or API issue it constitutes an undesired surprise to
users including MNE contributors.

> I'll try to write the fix before the release.

Yes that would be great.

kingjr, 2016-04-20:
@fraimondo I've just changed the default scoring approach. I suggest you write your ENH once #3143 is merged to avoid conflicts.

fraimondo, 2016-04-20:
I'm waiting for it.

kingjr, 2016-05-10:
@fraimondo can you close the issue?

closed 2016-05-11 by Eric89GXL