mne-tools/mne-python #3126: Can't import brainvision data

Pablo-Arias, 2016-04-14:
Hello all,
I'm having issues importing raw EMG data with mne.io.read_raw_brainvision(vhdr_fname, preload=True, eog = ()).
The same data is importing well in brainstorm software.

I think it's because my EMG data has these units (this is extracted from the my datas' header file):

Ch128=ZYGO,,1,C
Ch129=Corr,,1,µS
Ch130=131,,1,µV
Ch131=132,,1,ARU
Ch132=133,,1,µV

The problem might be that there are no conversion units in brainvision.py for C, µS....
Indeed, in brainvision.py there is only suport for these units : _unit_dict = {'V': 1., u'µV': 1e-6, 'uV': 1e-6}

And that's the log I have.

Thanks a lot,
Pablo

[image]

Pablo-Arias, 2016-04-14:
This is the header file I'm using:
[1.txt](https://github.com/mne-tools/mne-python/files/219340/1.txt)

Pablo-Arias, 2016-04-14:
When I change the header file to : 

Ch128=ZYGO,,1,V
Ch129=Corr,,1,µV
Ch130=131,,1,µV
Ch131=132,,1,V
Ch132=133,,1,µV

It works, so it's a problem of units, I think branvision.py should handle other units than V.

Eric89GXL, 2016-04-14:
I agree it should be changed to accommodate these new units if possible. Do you have time to try making a PR to fix it?

Pablo-Arias, 2016-04-18:
I just did the PR

jona-sassenhagen, 2016-04-20:
Did you do it vs. master? I don't see it under https://github.com/mne-tools/mne-python/pulls

Pablo-Arias, 2016-04-21:
Just did it again, it's called 'Added more units to brainvision.py io ', sorry, I'm new to github!

closed 2016-06-03 by agramfort