mne-tools/mne-python #3127: mne.viz.topomap sensor name size too small

moritzschaefer, 2016-04-14:
Names on the topomap are always printed small:
https://github.com/mne-tools/mne-python/blob/master/mne/viz/topomap.py#L604
`ax.text(p[0], p[1], ch_id, horizontalalignment='center', verticalalignment='center', size='x-small')`

That's too small if you choose a size of several inches (e.g. 10).

Possible solutions:
- Provide another API parameter 
- Derive the parameter from the 'size' parameter

jona-sassenhagen, 2016-04-20:
How about simply increasing the default?

I agree it's usually unreadable.

moritzschaefer, 2016-04-20:
That would be an easy solution to start with. Just feels a bit like a workaround

jona-sassenhagen, 2016-04-23:
Ping @agramfort @jaeilepp ?

I volunteer for a PR if you agree this is an issue, but we would have to decide on the API (vs. e.g. defaults) first.

agramfort, 2016-04-23:
if you can infer a good value from size param it's nice.

moritzschaefer, 2016-04-23:
+1

2016, 22:59:

jona-sassenhagen, 2016-04-24:
I won't get to it for .12 though.

dengemann, 2016-05-19:
Another solution, just edit the figure post-hoc:

```Python
for tt in plt.findobj(fig, matplotlib.text.Text):
    if tt.get_text() in evoked.ch_names:
        tt.set_fontsize(14)
```

dengemann, 2016-05-19:
In the latter case make sure to plot with show=False and call fig.canvas.draw() and plt.show() once you're done

Eric89GXL, 2017-02-14:
@jona-sassenhagen any time to tackle this one?

agramfort, 2017-07-22:
I upvote @dengemann 's answer. It should do the job.

closed 2017-07-22 by agramfort