mne-tools/mne-python #3128: Q? neuromag vectoriview layout naming

kingjr, 2016-04-14:
The neuromag layout called `Vectorview-mag`, but the connectivity is `neuromag306mag` (and similar for `grad` etc)

https://github.com/mne-tools/mne-python/tree/master/mne/channels/data/layouts

I found it a bit confusing. Can't we make the naming consistent?

agramfort, 2016-04-14:
these files are not ours. We just expose them.

kingjr, 2016-04-14:
Shouldn't we make a lookup table to impose consistency?

dengemann, 2016-04-15:
Not sure it's worth the effort, Fieldtrip users for example should already
be very familiar with what we have. But what I think would help indeed at
least would be an option to list layouts from the command line in an
informative, as these names are not necessarily easy to remember.

agramfort, 2016-04-15:
if you write the narrative doc on this I am ok to negotiate :)

kingjr, 2016-04-15:
We could also add a page somewhere in the manual to add a table listing the 2D, 3D and neighbours layout of each system we currently expose.

agramfort, 2016-04-15:
thanks for volunteering :)

kingjr, 2016-04-26:
crawling under work ATM, closing for now it's low priority.

closed 2016-04-26 by kingjr