mne-tools/mne-python #3130: Error in reading evoked fiff which is saved from CNT

cfoodc, 2016-04-15:
Hi all, 

I am recently trying to use MNE to deal with some CNT files.
It seems that there is a problem in CNT IO.
I can use evoked.save to save my processed CNT file but I can not use mne.Evoked() to read it.
The problem only happened when I was dealing with CNT file, other fiff files are totally fine.

I found that the possible cause might be due to the incomplete meas_info from CNT-fiff.
Even the info data of CNT is correct before I save it to fiff, it turns out there are no nchan, sfreq , and other crucial information when I used the mne.EvokedF() to read the fiff.
Thus, the following error will appear when I try to read the evoked fiff.

If you need the original CNT file to reproduce the error, I can upload it.
Hope the description is clear.

Best, 

CF

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-9-4e67f23afc3d> in <module>()
----> 1 mne.Evoked('/Users/cf/EEG/cnt_epochs_1-ave.fif')

//anaconda/lib/python3.5/site-packages/mne/evoked.py in __init__(self, fname, condition, baseline, proj, kind, verbose)

//anaconda/lib/python3.5/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    607         with use_log_level(verbose_level):
    608             return function(*args, **kwargs)
--> 609     return function(*args, **kwargs)
    610 
    611 

//anaconda/lib/python3.5/site-packages/mne/evoked.py in __init__(self, fname, condition, baseline, proj, kind, verbose)
    102         self.info, self.nave, self._aspect_kind, self.first, self.last, \
    103             self.comment, self.times, self.data = _read_evoked(
--> 104                 fname, condition, kind)
    105         self.kind = _aspect_rev.get(str(self._aspect_kind), 'Unknown')
    106         self.verbose = verbose

//anaconda/lib/python3.5/site-packages/mne/evoked.py in _read_evoked(fname, condition, kind)
   1157     with f as fid:
   1158         #   Read the measurement info
-> 1159         info, meas = read_meas_info(fid, tree, clean_bads=True)
   1160         info['filename'] = fname
   1161 

//anaconda/lib/python3.5/site-packages/mne/io/meas_info.py in read_meas_info(fid, tree, clean_bads, verbose)

//anaconda/lib/python3.5/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    607         with use_log_level(verbose_level):
    608             return function(*args, **kwargs)
--> 609     return function(*args, **kwargs)
    610 
    611 

//anaconda/lib/python3.5/site-packages/mne/io/meas_info.py in read_meas_info(fid, tree, clean_bads, verbose)
    818 
    819     if sfreq is None:
--> 820         raise ValueError('Sampling frequency is not defined')
    821 
    822     if len(chs) == 0:

ValueError: Sampling frequency is not defined

teonbrooks, 2016-04-15:
to read the evoked from disk, you should use `mne.read_evokeds`. Feel free to re-open this issue if that doesn't work for you

teonbrooks, 2016-04-15:
sorry, that's not the case. yeah, if you can share the file, that would be great.

agramfort, 2016-04-15:
I just bumped into a similar issue yesterday. To replicate my pb:

```
from mne import read_evokeds
from mne.datasets import sample

data_path = sample.data_path()

fname = data_path + '/MEG/sample/sample_audvis-ave.fif'

condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0),
                      proj=True)

evoked.info['chs'][9]['loc'] = None

evoked.save('test-ave.fif')
read_evokeds('test-ave.fif')
```

some of our readers don't populate all the necessary fields of chs.

cfoodc, 2016-04-15:
I have uploaded the cnt file to the following link and the .ipynd is also attached
Thanks,

[download here](https://drive.google.com/open?id=0B4zpukADF-gFZ0Y1Njd6U0JyM1E)

I have checked the meas_info of fiff I saved.
It seems that the meas_info is not updated after evoked.save() thus it can't get the correct meas_info when I read the fiff

jaeilepp, 2016-04-15:
``write_evoked`` seems to be extremely fragile to empty info fields. Fix on the way. In the mean time you can work around the issue by populating the empty fields of info (I think in your case 'last_name', 'first_name', 'description').
Thanks @cfoodc for pointing this out.

closed 2016-04-19 by Eric89GXL

cfoodc, 2016-04-20:
Thanks a lot.
It works right now.